bambulab/BambuStudio #3987: Bad flushing volume recalculation range

Bugsmell, 2024-05-01:
### Bambu Studio Version

Beta 1.9.1.58

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

8th gen Intel 8750h
32GB
Nvidia gtx

### Printer

Bambú Lab P1S

### How to reproduce

1. Go-to Flushing volumes.
2. Set multiplier below 1.
3. Flushing volumes are set to 0 unless the multiplier is greater than or equal to 1
4. Print results are weird without purging when changing the filament.
-----------
A. Open a project created with a previous (stable) lo version and with a flushing multiplier below 1
B. Automatically and hiddlenly flushing volumes are set to 0 even without filament changes in AMS or sinchronization.
C. Print results are weird without purging when changing the filament.

### Actual results

Not accurate color printing 

[image]

### Expected results

Have an accurate color printing, as usual.

### Project file & Debug log uploads

Can be reproduced with any project and without any data. You only need an AMS.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MklBambu, 2024-05-07:
Thank you for reporting this issue. Could you please let me know your region and language of use?

Bugsmell, 2024-05-07:
Hello!, I have tried in Spanish and English. The purge volume calculation only fails in Spanish and for multiplier values less than 1.00. The region is Europe.
Thank you.
[image]
[image]

MklBambu, 2024-05-07:
Thanks! We will fix it in the next version.

Bugsmell, 2024-05-07:
Thank you. Glad to help.

See86, 2024-05-07:
Hello, 
I have the same issue in French, seem to be only working in english

MklBambu, 2024-05-17:
Hi, the new version has been released. You can try it out and verify if any issues still exist.
https://github.com/bambulab/BambuStudio/releases/

tomaszpagacz, 2024-05-19:
[image]

Hi. For me the issue exists in version 1.9.1.67 on Mac with chip Apple M2 and OS Sonoma 14.4.1 (23E224) in English. Multiplier was never changed and it is suddenly 0. When changed to 1 it has correct values. There is no warning before printing without flushing between 2 different material which seem to be relatively easy and helpful implementation.