bambulab/BambuStudio #3992: AMS - Custom Filament Profiles Dissapearing

ClawzSkunk, 2024-05-02:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

AMD Ryzen 9 3950X
32GB memory
Nvidia RTX 4090

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Select device
Load Custom filament profile into AMS slot with PA profile
Wait 3 seconds
Filament profile and flow dynamics calibration are removed and replaced with blank slot and “?”

### Actual results

[image]
[image]
[image]

filament profile continuously removed despite saving.

### Expected results

Calibrated Profile remains until spool empty and replaced or changed by user manually.

### Project file & Debug log uploads

Project file is irrelevant as this happens before any project can be printed.

### Checklist of files to include

- [ ] Log file
- [X] Project file

MadNasedo, 2024-05-04:
I had the same problem here with custom filament profiles - after selecting, it takes ~10 sec. then the ? appears.

murongguao, 2024-05-06:
This issue is due to you creating a custom Filament and then setting it directly in the Device interface of Studio. There is currently a problem with doing this, which is that the printer does not have the information for this custom Filament, so it will set the Filament information you set to empty, which will become "?".
This issue has been resolved in the latest beta version of Studio and will need to be resolved in conjunction with the latest printer beta firmware.

Another solution is to confirm that cloud synchronization is enabled after creating a custom Filament, and then go to the printer to enter the Filament settings interface of AMS (this step is to have the printer obtain Filament data from the cloud). Open the dropdown menu to check if the custom Filament you created exists. If it exists, you can set it again in Studio without any problems.

DanBao-Bambu, 2024-05-06:
This issue is due to you creating a custom Filament and then setting it directly in the Device interface of Studio. There is currently a problem with doing this, which is that the printer does not have the information for this custom Filament, so it will set the Filament information you set to empty, which will become "?".
This issue has been resolved in the latest beta version of Studio and will need to be resolved in conjunction with the latest printer beta firmware.

Another solution is to confirm that cloud synchronization is enabled after creating a custom Filament, and then go to the printer to enter the Filament settings interface of AMS (this step is to have the printer obtain Filament data from the cloud). Open the dropdown menu to check if the custom Filament you created exists. If it exists, you can set it again in Studio without any problems.

ClawzSkunk, 2024-05-06:
Hi, I should have mentioned that the same thing happens when trying to select the filament direct from the printer. I can see the filament profile and select it, but it continually shifts back to the "?".

DanBao-Bambu, 2024-05-06:
@ClawzSkunk This is a problem that I have not encountered before. If possible, please export the printer's log and send it to my email. It is possible that the printer's log is very large. You can place it in the network drive and set access permissions so that I can access it.

OzDruiD, 2024-05-06:
@DanBao-Bambu  is this resolved in the latest Public release v1.8 thats now out, and if so - do we still need to use BETA v4 off Bambu Studio?

fsnuxer, 2024-05-06:
Hello. Same issue even with latest X1C firmware (01.08.00.00)  
( Filament slot information turn to "?" after a few seconds when choosing it in bambustudio or bamby handy. )
I can't see/choose the profile on the X1C screen....

User profile / sync seems ok since I can select a custom filament defined in bambu studio from bambu handy app.

X1C firmware version 01.08.00.00. (Don't know if relevant but it's an early Kickstarter version)
bambustudio 19.1.158 ([1.9.1 Public Beta 4] flatpak
latest bamby handy on android from 17/04

DanBao-Bambu, 2024-05-11:
@fsnuxer When the printer is unable to update data from the cloud, it does reset the settings of Studio and printer. You can try logging in to the printer again to see if this problem can be resolved.

DanBao-Bambu, 2024-05-11:
@OzDruiD This feature started in Beta, you can update and try it out now!

fsnuxer, 2024-05-12:
Thanks !

Yes. It's working now that I logged out/and in again on the printer.
Suggestion : mention it in the error message on the printer or in the
slicer when "?" appears on the filament name.
Or better yet, investigate why we have to log out and in again ;)

DanBao-Bambu, 2024-05-13:
@fsnuxer I am glad to help you. I have forwarded your suggestions to the relevant department.

DanBao-Bambu, 2024-05-15:
@ClawzSkunk Hello, will this issue still appear on your Studio?

mavrc, 2024-05-18:
Can confirm this has gotten worse: in the past I have been able to use custom filament profiles with the "load the settings directly on the printer" trick, but as of today I can no longer use custom filament profiles at all - they always reset after a few seconds. The custom filament profiles appear and are selectable on both softwares and the printer but always revert after a short wait.

X1C firmware 1.08, Studio 1.9.1.66, Handy 2.10

mavrc, 2024-05-20:
A link to the log will be coming to you shortly. 

I have two systems running Studio, both running the same version, but one Windows and one Linux. This problem has gotten even weirder in that my custom filaments, which I previously defined on my Windows system, seem to be working fine. However, the two new filament definitions I created on the Linux version of Studio do not work. I don't know if this is related to the OS or not, or if this has to do with a particular version of Studio instead, but I hope the printer log will help.

DanBao-Bambu, 2024-05-20:
@mavrc It is highly possible that the Studio on both of your systems is connected to the same printer at the same time?

mavrc, 2024-05-20:
It is entirely possible that I left Studio open on my Windows PC while using my laptop (Linux.)  And yes, they are both logged in to the same cloud account, and I have only one printer, so yes, they may be connected simultaneously. Would this cause the behavior I encountered?

DanBao-Bambu, 2024-05-20:
@mavrc Yes, if two studios are opened simultaneously and one studio creates a new custom Filament, the other printer will not be aware. So when the same printer is connected to two studios, the other studio that is unaware of this custom Filament will issue a reset command.

magic-happenz, 2024-06-13:
Can confirm the same behavior also happens when you try to adjust the name of the profile after its creation by editing the JSON files. Is there also a workaround for that? I was actually about to open a new issue to report that I think you should remove the naming enforcement that automatically attaches a "@Bambu Lab X1 Carbon 0.4 nozzle" after creation. Its a custom filament profile and I wish to have m own naming to it. It also makes the names unnecessary long as compatible printers and nozzles are set in the profile already. How should I proceed with this?

DanBao-Bambu, 2024-06-19:
@magic-happenz Actually, I don't recommend you directly edit the JSON file in the file system. If you do this, it is highly likely that it will be deleted by cloud synchronization. Didn't Studio remove the suffix when displaying custom materials now? This suffix is meaningful for whether the printer supports this custom material, so don't delete it.

magic-happenz, 2024-06-20:
@DanBao-Bambu yeah I get that, I just tried everything I could as Bambu Studio adds the suffix and does not allow to remove it. However, adjusting the JSON files does not delete them by cloud synchronization but kinda breaks them to be synced to the AMS. I wish Studio removed those suffix from everywhere but its not the case. Running version 1.9.2.57 it still shows up in the filament settings: 
[image]

DanBao-Bambu, 2024-06-26:
@magic-happenz The suffix here should not be removed because this interface has two entrances, one from the preparation page and the other from the edit button of the custom material. If you enter from the editing entrance of the custom material interface, it is very likely that the printer corresponding to the preset is not the same as the printer you currently selected on the preparation interface. If the suffix is removed, it will be unclear which preset it was edited for.

magic-happenz, 2024-06-26:
I see. Thank you for the explanation.