bambulab/BambuStudio #3993: Windows 11 automatic tray hide

poli27, 2024-05-02:
### Bambu Studio Version

1.9.1.58

### Where is the application from?

Bambu Lab github releases

### OS version

win 11

### Additional system information

AMD Ryzen 9 7950X 16-Core Processor   4.50 GHz
 32,0 GB ram
AMD RX7800XT VGA

### Printer

P1P/S

### How to reproduce

automatic tray hiding, if the bambu studio is open, the tray cannot be brought up!

### Actual results

the Bambu studio must be reduced in size, and the tray can only be accessed afterwards!

### Expected results

Everything should be accessible to the tray.. there was no such error in the old versions!

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-05-04:
thanks for reporting this issue
could you give a snapshot of it?

poli27, 2024-05-04:
> thanks for reporting this issue could you give a snapshot of it?

https://www.youtube.com/watch?v=_fZdxO0eKtU

MadMax389, 2024-05-05:
I also reported this issue back in December #3094

MadMax389, 2024-05-06:
Ok no problem, it's only a minor inconvenience. I just run it windowed.

lanewei120, 2024-08-08:
according to https://github.com/bambulab/BambuStudio/issues/3993#issuecomment-2095043599
close it

closed 2024-08-08 by lanewei120