bambulab/BambuStudio #3996: Perpendicular holes add thickness to inner surface where the holes are.

therobinojones, 2024-05-02:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 and Mac OS Ventura

### Additional system information

Processor: AMD Ryzen 9 7950X 16-Core Processor
Ram: 64GB
Video: RTX 4080

This bug also appears on my MAC M1 Laptop. 

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Load the attached project, labeled holes_bug.3mf
2. Slice the project
3. Observe the objects with the perpendicular holes where the inner wall are plugged vs open.

### Actual results

You will see a band around the area of the holes on the throughout holes on the objects without wall plugs. The objects with plugged walls are ok with even surface. 

### Expected results

When slicing an object that has through holes, the inner face should not add any thickness. This is very annoying when designing precise slip fit objects. I was able to find a work around by plugging the walls with a very thin wall which the slicer opens up because it's too thin, and this fixes the issue.

You can read more about the issue on the official bambu lab facebook group page here:

https://www.facebook.com/groups/bambulabofficialusers/posts/1181573149453369/?comment_id=1181629692781048&notif_id=1714687442204590&notif_t=feedback_reaction_generic&ref=notif

### Project file & Debug log uploads

[holes_bug.zip](https://github.com/bambulab/BambuStudio/files/15194686/holes_bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

therobinojones, 2024-05-03:
****Important Update:

I discovered that the issue is introduced when using 'X-Y hole compensation' and 'X-Y contour compensation. ' individually or both. Setting both to 0 (the default) it slices without issues.

QingZhangBambu, 2024-05-06:
Contours and holes are judged in two dimensions, which can lead to errors. for instanse: 
[image]
[image]

It is a difficult problem to solve. We'll put it on the list and try to find a solution.