bambulab/BambuStudio #3997: A1 Mini - light always off.

NVNDO, 2024-05-03:
Problem: It's hugely annoying having to manually switch off the light every time one starts the printer. 

Solution: Implement an option to switch off the light permanently.  

Alternative: In addition to the existing Light on/off button, add
Light on startup on/off function.

NVNDO, 2024-05-06:
> @NVNDO in the meantime I think you can do it here:
> 
> 
> and on the printer itself and app. But still, it's switching them on and having to switch off each printer manually. I have 2 right now but planning on more in a near future, and I'm not willing to do this every day. 

Nor do other people.. the community forum has many people complaining about this, but no one has yet come here to upvote (which doesn't help)

ArnieO, 2024-05-08:
I totally agree, this should be a parameter that could be set to the opposite default setting somewhere in Bambu Studio.

LTB3D, 2024-05-11:
Same issue here. It's so annoying!

Alpha12o, 2024-05-14:
Same. Every time I start a print I have to open the app or toggle the light off on the printer touchscreen.
I would like for it to be a setting you can enable/disable, like the sound. IMHO, the default behavior should be printing with light off and if you want, you open the app or use the printer touchscreen to switch the light on.

NVNDO, 2024-05-15:
> Same, but in my case with A1. Every time I start a print I have to open the app or toggle the light off on the printer touchscreen. I would like for it to be a setting you can enable/disable, like the sound. IMHO, the default behavior should be printing with light off and if you want, you open the app or use the printer touchscreen to switch the light on.

Hey @Alpha12o , I didn't know this was also a thing on the A1s, therefore I changed the title to extend this petition. 
thx for joining!

ArnieO, 2024-05-15:
> I didn't know this was also a thing on the A1s

I confirm: Same issue on A1. Which is why I think this should be an option in Bambu Studio (light ON or OFF), when sending the job to the printer.

GenericArt, 2024-05-15:
I also would like to see the ability to permanently turn off the light as it is super annoying to do it every print.

Malrikgus, 2024-05-20:
Upvoted

Jacques44, 2024-05-25:
Hello, upvote !

jonathanoeijoeng, 2024-05-25:
Currently I use beta firmware on A1 mini and realized that it does not turn on light every print.

ArnieO, 2024-05-25:
> Currently I use beta firmware on A1 mini and realized that it does not turn on light every print.

Where is the beta firmware available?

jonathanoeijoeng, 2024-05-25:
You need to enable beta firmware on bambu handy apps first.

NVNDO, 2024-05-25:
@jonathanoeijoeng and then..? Where does one get the firmware from? In the App it says, " You joined the beta-program" but from what I could read beforehand, one must be accepted first to be able to use the beta. 

Do they send a confirmation like "You've been accepted"? 
Does it downloads automatically, does one have to do it manually?

jonathanoeijoeng, 2024-05-25:
I didnt get any confirmation but need to wait a couple of days before new
beta firmware showed up in the printer.

MarcoQ3D, 2024-05-27:
Good evening, thanks for this post. I would also prefer to have the option to choose the light off when the A1 starts. I follow

Alpha12o, 2024-05-27:
I also signed up for the beta firmware program.
I guess we'll have to wait for a new beta update

NVNDO, 2024-08-26:
Seems like they just switched from ON to OFF, still NOT giving us the option to choose!

Pretty frustrating at this point, as who owns a P1S as well might want to have the light ON when switching the printer ON because its closed and you don't see a thing in there otherwise. Whereas the A1 or A1 Mini is open (no case around it) and therefore does NOT need a light switched on at startup. Either way - opinions may vary . that's why we need an OPTION TO CHOOSE FROM.

Can someone from Bambulabs PLEASE implement this?! @XunZhangBambu 
"Light at Startup: ON / OFF"

At this point I'm just hoping - I'll thank later when its done.

jonathanoeijoeng, 2024-08-26:
I use the latest firmware and lamp still turn off after I turn printer on
or start to print.
It already implemented couple months ago

NVNDO, 2024-08-26:
@jonathanoeijoeng You did not properly read my post. Please read again.

jonathanoeijoeng, 2024-08-26:
Ok then

esmith13, 2024-08-27:
I have this issue on my brand new P1S. Light comes on when I power the printer on. If I manually switch it off in Bambu Studio during a print, it still switches back on the next print - regardless if I power cycled the printer or not (I typically power on once and do many short prints - light comes on again each print).

I would love to keep the light OFF as I have my own lighting solution.

NVNDO, 2024-08-27:
@esmith13 Hey there. How come you have the exact opposite problem? ^^  wish the lights came on when switching on the printer, as prior to printing I usually change or clean the buildplate and can not see sh... without light. It does come on once I start a print, but then its too late. 

For now I have mitigated the issue by unplugging my lights from the board and connecting it to the USB port. That way, as soon as I start the printer - the light goes on. But it does of course never go off, unless I switch off the printer - so I's still like this to be implementes as my approach is more a workaround than a solution.

If the light bothers you, and you have light from an external source, just unplug the connector inside the printer.

esmith13, 2024-08-27:
> If the light bothers you, and you have light from an external source, just unplug the connector inside the printer.

Yeah, I could do that, but I would hope for a proper solution. Can't hurt to ask & hope - even if I wind up unplugging it in the meantime. 

Thanks.