bambulab/BambuStudio #3998: Frequent Application Hang with Large Projects - MacOS Pinwheel

infinityends, 2024-05-03:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 14.4.1

### Additional system information

MacBook Pro 16" M1 Max 64GB RAM 2TB SSD

### Printer

X1 Carbon

### How to reproduce

Open a large project file. I've run into this with multiple large files from maker world, for example the catan board models.
https://makerworld.com/en/models/449350

Try to delete an entire print plate from the file, program hangs.

### Actual results

Program hangs with spinning pinwheel - activity monitor shows 100% cpu usage - application must be force closed, even after letting the computer sit for 5+ minutes it is still hung

### Expected results

plate should delete, even if program sits for a short time to work through it.

### Project file & Debug log uploads

https://makerworld.com/en/models/449350
[Archive.zip](https://github.com/bambulab/BambuStudio/files/15196280/Archive.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

infinityends, 2024-05-03:
I installed the latest 1.9.1.58 beta4 build tonight and it seems like this performance issue does not exist in the beta app so it is likely resolved for the next public release.

lanewei120, 2024-06-12:
from above comment, close it as not appeared on public release

closed 2024-06-12 by lanewei120