bambulab/BambuStudio #4000: Change "Combinate" to "Combine"

MadMax389, 2024-05-03:
### Bambu Studio Version

1.9.1.58

### Where is the application from?

Bambu Lab github releases

### OS version

Win11

### Additional system information

10th Gen Core i5

### Printer

Slicer - All Printers

### How to reproduce

The word "Combinate", in English, is a noun meaning the result of a combination. A verb like "Assemble" (as it was previously), "Combine", or "Group" would be more appropriate in the context menu in the slicer. Minor issue, I know, but "combinate" doesn't make sense in this context.
[image]

### Actual results

NA

### Expected results

NA

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-05-06:
Hello, after our internal discussion, we plan to use the same word as prusaSlicer for the word "combine": Merge.
This issue will be resolved in the upcoming OTA version~

scotta316, 2024-05-10:
"Merge" sounds permanent, like a Boolean union. Assemble makes more sense, as there will still be individual parts.

Haidiye00, 2024-05-10:
It's a bit, but we still refer to the term prusa

Haidiye00, 2024-05-14:
This issue has been fixed. Please use the latest version. Thank you

closed 2024-05-14 by Haidiye00