bambulab/BambuStudio #4004: Printing stopped because nozzle temperature problem

amyreeb, 2024-05-05:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lap 

### How to reproduce

Any print that I start or print calibration that I start results in the same error message: Printing stopped because nozzle temperature problem 0300 8008 16:19:26

My printer was working until I got a glob of PLA stuck on the hotend. This required me to replace the Hotend with a hotend hardend steel nozzle 0.4mm (I was also previously using a 0.4mm). I also replaced the silicone sock. After installing and then getting this error message any time I try to print - I disconnected and reconnected everything again to see if any connections were loose but they were not. I also turned off, unplugged, replugged power, and turned back on for a full restart to test and that didn't help either.
[image]

Please help - as right now I am completly unable to print anything

### Actual results

[image]

### Expected results

I should be able to print without an error messages

### Project file & Debug log uploads

[Dragon dice tower.zip](https://github.com/bambulab/BambuStudio/files/15214654/debug_network_Fri_May_03_12_24_28.log.zip)
ps://github.com/bambulab/BambuStudio/files/15214650/Dragon.dice.tower.zip)

[debug_network_Fri_May_03_12_24_28.log.zip](https://github.com/bambulab/BambuStudio/files/15214655/debug_network_Fri_May_03_12_24_28.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

amyreeb, 2024-05-05:
Printer should say Bambu Lap P1P

amyreeb, 2024-05-06:
Any help on this? my entire printer doesn't work right now due to this error. Impossible to print anything

mrunion, 2024-05-06:
You should really be asking Bambu Lab Support and not the Bambu Studio code repository. You should be able to find information on what to check for that error message here:

https://wiki.bambulab.com/en/hms/home

shuizhiguo, 2024-05-09:
maybe you can know this link  :
https://wiki.bambulab.com/en/p1/maintenance/complete-hot-end-assembly
Perhaps it's a hardware issue, please check the plug or contact after-sales service， thank you

amyreeb, 2024-05-09:
How do I contact after sales service?