bambulab/BambuStudio #4008: Thin items not rendering in Z correctly

AbeFM, 2024-05-07:
### Bambu Studio Version

18451

### Where is the application from?

Bambu Lab Official website

### OS version

win11

### Printer

X1C AMS

### How to reproduce

Load my 3MF
Slice

Oddly, I've managed to get this to slice right later and can't tell what I changes

### Actual results

I have tried to recreate metal 'screen', but to make it easy to print, it's a bunch of X-axis 0.5mm wide lines which are 0.5mm thick from a roof above them, and another set perpendicular which are half the height (again, 0.25 mm in Z).

I chose the width to render in 1 extrusion, and the height just a bit over one layer line to help the slicer "see" them.

I kept getting stubs even though my design is flat in Z:
[image]

You can see how the nubs on the layer below and the crosses on the top rendered layer seem to indicate the layers are misalligned. My feature is 1mm from the surface with 0.2 layers and 0.2 first layer, so this shouldn't be. I can vary the first layer, but constantly get some variation of an incorrect rendering - INCLUDING a setting where there's only one layer with any screen at all, even though the model is 0.5 or 2.5 layer thick at that point.

I'm not sure what is causing this, but it can make some of these designs difficult when I cannot intentionally make something 1 layer thick or leave a 1 layer gap.

### Expected results

Should look like this:
[image]
Threads going one direction, then on the next layer, bridges between them.

### Project file & Debug log uploads

[Support Missing Feature Issue.zip](https://github.com/bambulab/BambuStudio/files/15227712/Support.Missing.Feature.Issue.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-05-20:
Hello, could you record a GIF or video explaining the operation and issues? Thank you

AbeFM, 2024-05-21:
I'm not sure how a video would help?

Look at the model. The objects should fall within one z-layer but they take up several. Perhaps a more simple model would help?

Haidiye00, 2025-06-11:
https://github.com/bambulab/BambuStudio/releases/tag/v02.01.00.59
Hello, please use the latest version, thank you.

[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)
I'll close the question for now. If you have any further questions, let's continue the discussion. If there are still issues, I suggest recording a GIF. Thank you