bambulab/BambuStudio #4015: Simplify Mesh Bug

dsopala, 2024-05-07:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

x1c

### How to reproduce

When you are in the simplify menu, with the decimal ratio radial button pressed, typing in a value does not work. The slider works great, but when you type in a value it does nothing to the mesh.

I tried hitting Tab, Enter, and Appy, but it does not work.

### Actual results

nothing happens

### Expected results

the mesh simplifies. 

### Project file & Debug log uploads

I don't have one

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2024-06-03:
The issue can be confirmed. I will fix this issue.
Thanks for your report !

MackBambu, 2024-10-18:
Fixed

closed 2024-10-18 by MackBambu