bambulab/BambuStudio #4017: Bambu Stuber Slicer Scale arrows not correct

PoliticWatch, 2024-05-08:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 PRO

### Printer

Bambu X1 Combo

### How to reproduce

1. Click Object on Plate to resize
2. Click Scale Icon on top menu
3. Arrows on object do not match up when changing % vs actual size
[image]
[image]

### Actual results

You will notice that the arrows for Y and Z point in different directions.

### Expected results

Expect that the arrows would point in the correct axis

### Project file & Debug log uploads

Any file

### Checklist of files to include

- [X] Log file
- [X] Project file

PoliticWatch, 2024-05-08:
That is % and Size for each of Y and Z axis point in different direction.  Z is the example in pictures.

Haidiye00, 2024-05-08:
Hello, I just noticed that the latest version of V1.9 does not have this issue. Can you confirm it?

PoliticWatch, 2024-05-10:
Checking for updates, it says I have the latest version.

Do I need to manually download somewhere?

- John

Haidiye00, 2024-05-10:
Of course ,you can also download from https://github.com/bambulab/BambuStudio

PoliticWatch, 2024-05-10:
I did not see a Windows Executable file.  Seems I will need to learn how to compile.  Will look this week end.

Haidiye00, 2024-05-10:
[image]
https://github.com/bambulab/BambuStudio/releases/tag/v01.09.01.58

Haidiye00, 2024-05-14:
This issue has been fixed. Please use the latest version. Thank you

closed 2024-05-14 by Haidiye00

PoliticWatch, 2024-05-16:
Thank you.  Downloaded…

 

Closed #4017 <https://github.com/bambulab/BambuStudio/issues/4017>  as completed.