bambulab/BambuStudio #4018: Building from source fails on Fedora 40

LightTreasure, 2024-05-08:
### Bambu Studio Version

v01.09.01.58

### Where is the application from?

Built from source by yourself or others

### OS version

Fedora 40, Linux 6.8.8-300.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Sat Apr 27 17:53:31 UTC 2024 x86_64 GNU/Linux

### Additional system information

gcc version 14.0.1 20240411 (Red Hat 14.0.1-0) (GCC)
cmake version 3.28.2
GNU Make 4.4.1 Built for x86_64-redhat-linux-gnu

### Printer

Bambu Lab X1C

### How to reproduce

1. Clone this repository on Fedora 40 with the latest system updates
2. Follow the steps in the [compile guide](https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide)

### Actual results

After executing the `cmake --build . --target install --config Release -j1` command, I get the following compile-time error:

```
[  3%] Built target clipper
[  3%] Building CXX object src/clipper2/CMakeFiles/Clipper2.dir/Clipper2Lib/src/clipper.engine.cpp.o
In file included from /home/alok/src/BambuStudio/src/clipper2/Clipper2Lib/include/clipper2/clipper.engine.h:22,
                 from /home/alok/src/BambuStudio/src/clipper2/Clipper2Lib/src/clipper.engine.cpp:16:
/home/alok/src/BambuStudio/src/clipper2/Clipper2Lib/include/clipper2/clipper.core.h:117:26: error: template-id not allowed for constructor in C++20 [-Werror=template-id-cdtor]
  117 |         explicit Point<T>(const Point<T2>& p) { Init(p.x, p.y); }
      |                          ^
/home/alok/src/BambuStudio/src/clipper2/Clipper2Lib/include/clipper2/clipper.core.h:117:26: note: remove the ‘< >’
cc1plus: all warnings being treated as errors
gmake[2]: *** [src/clipper2/CMakeFiles/Clipper2.dir/build.make:76: src/clipper2/CMakeFiles/Clipper2.dir/Clipper2Lib/src/clipper.engine.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:699: src/clipper2/CMakeFiles/Clipper2.dir/all] Error 2
gmake: *** [Makefile:166: all] Error 2
```

g++ is throwing a C++20 error, even though src/clipper2/CMakeLists.txt explicitly specifies C++17:

```
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS OFF)
```

### Expected results

Compilation should complete without errors

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [X] Log file
- [X] Project file

LightTreasure, 2024-05-08:
I investigated further. CMake is doing the right thing - it adds a `-std=c++17` flag for the compiler. What's happening is that gcc detects the `template-id-cdtor` issue and throws a warning. However, clipper2's CMakeLists.txt file passes the `-Werror` flag to gcc, which causes this warning to turn into an error. 

Adding `-Wno-template-id-cdtor` flag to clipper2's `target_compile_options` gets past that error

However, now I'm running into another build issue (Only showing the error here for clarity. If you need the whole build log, please let me know.):

```
/home/alok/src/BambuStudio_dep/usr/local/include/boost-1_78/boost/lexical_cast/detail/converter_lexical.hpp:243:13: error: static assertion failed: Target type is neither std::istream`able nor std::wistream`able
  243 |             BOOST_STATIC_ASSERT_MSG((result_t::value || boost::has_right_shift<std::basic_istream<wchar_t>, T >::value),
      |             ^~~~~~~~~~~~~~~~~~~~~~~
/home/alok/src/BambuStudio_dep/usr/local/include/boost-1_78/boost/lexical_cast/detail/converter_lexical.hpp:243:13: note: ‘(((bool)boost::integral_constant<bool, false>::value) || ((bool)boost::integral_constant<bool, false>::value))’ evaluates to false
gmake[2]: *** [src/libslic3r/CMakeFiles/libslic3r_cgal.dir/build.make:76: src/libslic3r/CMakeFiles/libslic3r_cgal.dir/MeshBoolean.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:994: src/libslic3r/CMakeFiles/libslic3r_cgal.dir/all] Error 2
gmake: *** [Makefile:166: all] Error 2
```

lanewei120, 2024-05-10:
@LightTreasure 
yes, please provide the full building logs from beginning

LightTreasure, 2024-05-10:
@lanewei120 Here are the full logs (Note that this is the output after adding `-Wno-template-id-cdtor` flag to clipper2's `target_compile_options`. If you would like the logs without adding that flag, I can add that too, but it's just the same output as I provided in the issue report at the top).

Output of `cmake .. -DSLIC3R_STATIC=ON -DSLIC3R_GTK=3 -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH="/home/alok/src/BambuStudio_dep/usr/local" -DCMAKE_INSTALL_PREFIX="../install_dir" -DCMAKE_BUILD_TYPE=Release`: [cmake_cmd_output_afterclipperchange.log](https://github.com/bambulab/BambuStudio/files/15270001/cmake_cmd_output_afterclipperchange.log)

Output of `cmake --build . --target install --config Release -j1`: [build_after_clipper_change.log](https://github.com/bambulab/BambuStudio/files/15270002/build_after_clipper_change.log)

lanewei120, 2024-05-11:
I am not sure what is the detailed reason for I can not reproduce it on my fedora 37, ubuntu 22.04
from the log
[image]
it seems related with MeshBoolean.cpp: 695?

how about change 
for (int n = 0; n < numConnComps; ++n) {
to 
for (uint32_t n = 0; n < numConnComps; ++n) {
for a try?

LightTreasure, 2024-05-11:
@lanewei120 Unfortunately, that didn't fix the issue. This is because the error is unrelated to that file. g++ is being stupid and carrying over the filename and line number from a previous warning (it is shown just before the top line in your screenshot):

```
/home/alok/src/BambuStudio/src/libslic3r/MeshBoolean.cpp: In function ‘bool Slic3r::MeshBoolean::mcut::do_boolean_single(McutMesh&, const McutMesh&, const std::string&)’:
/home/alok/src/BambuStudio/src/libslic3r/MeshBoolean.cpp:695:23: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
  695 |     for (int n = 0; n < numConnComps; ++n) {
      |                     ~~^~~~~~~~~~~~~~ 
```

... so making the change you suggested removes that warning, but it doesn't fix the error.

> I can not reproduce it on my fedora 37, ubuntu 22.04
Yup, I looked at the CI build log on github and it seems to be fine as well. I think this is an issue that started to show up because of some build tool upgrade after fedora 37. Maybe gcc or cmake? What is the version of cmake and gcc on your setup?

lanewei120, 2024-05-11:
it seems fedora 36 not fedora 37, sorry for the wrong information before

Linux version 5.17.5-300.fc36.x86_64 (mockbuild@bkernel01.iad2.fedoraproject.org) (gcc (GCC) 12.0.1 20220413 (Red Hat 12.0.1-0), GNU ld version 2.37-24.fc36) #1 SMP PREEMPT Thu Apr 28 15:51:30 UTC 2022

[lane.wei@fedora bamboo_slicer]$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/12/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,objc,obj-c++,ada,go,d,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --enable-libstdcxx-backtrace --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl=/builddir/build/BUILD/gcc-12.2.1-20221121/obj-x86_64-redhat-linux/isl-install --enable-offload-targets=nvptx-none --without-cuda-driver --enable-offload-defaulted --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux --with-build-config=bootstrap-lto --enable-link-serialization=1
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 12.2.1 20221121 (Red Hat 12.2.1-4) (GCC)

[lane.wei@fedora bamboo_slicer]$ cmake --version
cmake version 3.25.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).

LightTreasure, 2024-05-11:
Thanks! This is the output from my machine:

gcc:

```
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/14/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,objc,obj-c++,ada,go,d,m2,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --enable-libstdcxx-backtrace --with-libstdcxx-zoneinfo=/usr/share/zoneinfo --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl=/builddir/build/BUILD/gcc-14.0.1-20240411/obj-x86_64-redhat-linux/isl-install --enable-offload-targets=nvptx-none,amdgcn-amdhsa --enable-offload-defaulted --without-cuda-driver --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux --with-build-config=bootstrap-lto --enable-link-serialization=1
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 14.0.1 20240411 (Red Hat 14.0.1-0) (GCC)
```

And cmake:

```
cmake version 3.28.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

lanewei120, 2024-05-11:
is it possible to find out which line caused this error on your side?

LightTreasure, 2024-05-11:
I tried digging through the code and figuring this out. However, Boost code is notoriously complicated, even more than usual c++ code. I'll keep trying, but I'm hoping to find the solution here from people who know this code better than I do.

LightTreasure, 2024-05-29:
I found some ways to get past these (and other) errors and was able to successfully compile Bambu Studio on Fedora 40. These are hacky and generally bad ways to make things work **just for my system**. The Bambu devs need to provide more solid fixes.

Most of the compilation issues stem from gcc and cmake using C++20 or higher. Reverting to C++17 gets past them. Some CMakeList files already specify that C++17 should be used, but for whatever reason the cmake version I'm using (3.28.2) ignores those directives. There is probably a newer, "proper" way to specify the C++ standard in this version of CMake, and the Bambu devs need to fix that.

For my purposes, I resorted to explicitly specifying the C++ version by using the `add_compile_options()` cmake directive to directly send the `-std=c++17` compiler flag to gcc.

Here are the changes I made:

1. Installed the development package for libquadmath (for Fedora this is `libquadmath-devel`).   This is not specified in the list of dependencies on the Build Guide, and should be added there.
2. Clipper2 compilation fails due to a C++20 warning that is treated as an error. Removing the `-Werror` flag for Clipper2 compilation avoids this issue:
```
diff --git a/src/clipper2/CMakeLists.txt b/src/clipper2/CMakeLists.txt
index 275f9273..f22c3e24 100644
--- a/src/clipper2/CMakeLists.txt
+++ b/src/clipper2/CMakeLists.txt
@@ -37,7 +37,7 @@ target_include_directories(Clipper2
 if (WIN32)
   target_compile_options(Clipper2 PRIVATE /W4 /WX)
 else()
-  target_compile_options(Clipper2 PRIVATE -Wall -Wextra -Wpedantic -Werror)
+  target_compile_options(Clipper2 PRIVATE -Wall -Wextra -Wpedantic)
   target_link_libraries(Clipper2 PUBLIC -lm)

 endif()
```
3. libslic3r compilation errors are fixed using the c++17 flag:
```
diff --git a/src/libslic3r/CMakeLists.txt b/src/libslic3r/CMakeLists.txt
index ccf30d84..b8662c30 100644
--- a/src/libslic3r/CMakeLists.txt
+++ b/src/libslic3r/CMakeLists.txt
@@ -12,6 +12,7 @@ if (MINGW)
     add_compile_options(-Wa,-mbig-obj)
 endif ()

+add_compile_options(-std=c++17)
 set(OpenVDBUtils_SOURCES "")
 if (TARGET OpenVDB::openvdb)
     set(OpenVDBUtils_SOURCES OpenVDBUtils.cpp OpenVDBUtils.hpp)
```
4. libnest2D compilation errors are also fixed using the c++17 flag, and I think additionally the -mfloat128 flag to enable the gcc support for 128-bit floats.

```
diff --git a/src/libnest2d/CMakeLists.txt b/src/libnest2d/CMakeLists.txt
index c18dc31c..341b3808 100644
--- a/src/libnest2d/CMakeLists.txt
+++ b/src/libnest2d/CMakeLists.txt
@@ -22,7 +22,7 @@ set(LIBNEST2D_SRCFILES
     )

 add_library(libnest2d STATIC ${LIBNEST2D_SRCFILES})
-
+add_compile_options(-std=c++17 -mfloat128)
 target_include_directories(libnest2d PUBLIC ${CMAKE_CURRENT_LIST_DIR}/include)
 target_link_libraries(libnest2d PUBLIC NLopt::nlopt TBB::tbb Boost::boost libslic3r)
 target_compile_definitions(libnest2d PUBLIC LIBNEST2D_THREADING_tbb LIBNEST2D_STATIC LIBNEST2D_OPTIMIZER_nlopt LIBNEST2D_GEOMETRIES_libslic3r)
```

LightTreasure, 2024-05-29:
Found a way to properly specify c++17 in cmake options. PR #4172 has the changes.

PR #4172 (2024-07-13): Explicitly specify C++17 in libslic3r and libslic3r_cgal cmake, fixing build on Fedora 40
Bambu studio currently fails to build on Fedora 40. This is because the CMake configuration of `libslic3r` and `libslic3r_cgal` currently does not explicitly specify the C++ standard, but it seems like the `libslic3r` and `libslic3r_cgal` code assumes C++17. Without explicit specification, the compiler uses C++20.

Also added `-Wno-error=template-id-cdtor` to the compiler flags for `Clipper2`, preventing this warning from being thrown as an error. The `template-id-cdtor` warning being thrown as an error i [... cut]
closed 2024-07-13 by lanewei120