bambulab/BambuStudio #4021: hotend wire came off

rob74gen, 2024-05-09:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

rob74gen, 2024-05-09:
The hotend wire came off an its not working at all i have owned this sense feb or march

lanewei120, 2024-05-10:
> please call me 8164903511

please file a ticket from here https://bambulab.com/en/my/support/tickets/create
@rob74gen