bambulab/BambuStudio #4022: 版本1.8.4.51切片软件在联想thinkbook16+ 运行时加载视频会闪退软件，无法正常加载视频。

xubintww, 2024-05-09:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

windows11

### Additional system information

处理器：Intel core ultra 7 155H
内存：32GB
显卡：inter Arc Graphics

### Printer

X1c

### How to reproduce

1，打开软件
2，切片打印
3，转到设备摄像机加载视频，软件闪退。

### Actual results

提示视频加载中，等待10秒左右时间软件自动闪退

### Expected results

软件闪退

### Project file & Debug log uploads

文件大小过大：允许 25 MB，72 MB 尝试上传。

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2024-05-22:
依赖系统的 Windows Media Player 可能有问题。
参考 wiki 解决
https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view
我们将重构播放器，不依赖 WMP。

fahaaaxiki, 2024-12-06:
您好，我们在最新的版本：[1.10.1 Public Release](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50),中解决了这个问题，您可以升级到这个版本上使用并查看liveview功能是否正常。如果依然有问题的话欢迎再次反馈给我们。