bambulab/BambuStudio #4023: Add teardrop cylinder as negative part you can add

keithel, 2024-05-09:
## Is your feature request related to a problem?
I'm always frustrated when screwholes made in parts, when printed parallel to the bed, will not print well due to overhangs.
To solve this, I can make the screwholes teardrop shaped, rather than just plain cylinders, so that the angle of printing never is too steep that there is a layer that is not supported by the prior layer.

## Description of feature
Add a "teardrop cylinder" to  the choices of basic parts you can add when adding a negative part.

When I say "teardrop cylinder" I am meaning the union of a cylinder and... well, I guess a diamond with sides of length the same as the cylinders radius.

## Workarounds
The workaround for this is to add the screwholes using a modeling tool outside of the slicer.
Since this is very FDM printing-specific, I think it's a feature that makes sense to have in the slicer.

altruistically-minded, 2024-05-11:
A quick work around is to "Add Negative Part"->"Load" and then load an STL of a teardrop cylinder.

keithel, 2024-08-27:
Quickest is this, but should be easy to create from scratch:

https://www.printables.com/model/769742-negative-volume-double-tear-romboid-cylinder-for-h