bambulab/BambuStudio #4025: Text position reset upon edit

GuyH77, 2024-05-09:
### Bambu Studio Version

1.9.1.58

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 pro

### Printer

Bambu Lab X1C 0.4 nozzle

### How to reproduce

Add text to an object. Use the positioning handle to roughly place the correct location. Modify the text using the move command, to position accurately. Edit text, change text. The position resets based on the placement handle.

**Create Text**
[image]

**Position Text**
[image]

**Edit Text**
[image]

### Actual results

The text position resets upon text edit. The text tool doesn't allow precise placement so we use the move tool to position where we want. Upon edit the position is reset and we have to move the text again. Very frustrating. 

### Expected results

Test should persist and remain in the place we move it to. The location handle should only be to select the location surface and approximate position. 

### Project file & Debug log uploads

[TEXT_POSITION_TEST.zip](https://github.com/bambulab/BambuStudio/files/15265762/TEXT_POSITION_TEST.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2024-05-14:
hello,the problem has been identified and is currently being resolved

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00