bambulab/BambuStudio #4026: Firmware Update not possible in LAN mode

probonopd, 2024-05-09:
### Bambu Studio Version

Any

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

A1 mini

### How to reproduce

1. Put the A1 mini into LAN mode
2. In Bambu Studio, go to "Update"

### Actual results

[image]

### Expected results

I can update the printer firmware

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2024-05-13:
@probonopd 
In lan only mode, the printer cannot connect to the server and cannot obtain update information. If you want to upgrade the printer, you must turn off the lan only mode.

probonopd, 2024-05-13:
Thanks @walterwongbbl. I was hoping that Bambu Studio could download the update file, and upload it to the printer (similar to how it uploads gcode). This way, the printer would never need to have Internet access. In many environments (e.g., factories), machines must not access the Internet for security reasons.

As a fallback, it should be possible to put the update file on a microSD card and flash it that way.

Triskae, 2024-07-17:
The idea of upgrading the printer with a file (pushed from the slicer to the printer, or dropping the update file to the sdcard) would a really nice addition !!

SaltWei, 2024-07-30:
> I was hoping that Bambu Studio could download the update file, and upload it to the printer (similar to how it uploads gcode).

Conflict with https://github.com/bambulab/BambuStudio/issues/1429

probonopd, 2024-08-05:
> Conflict with https://github.com/bambulab/BambuStudio/issues/1429

So, we need a way to manually update the firmware without the printer needing to be connected to the cloud. E.g. by manually placing it on microSD card.

katherine-black, 2024-08-10:
> > Conflict with #1429
> 
> So, we need a way to manually update the firmware without the printer needing to be connected to the cloud. E.g. by manually placing it on microSD card.

Yea, I think it's totally not unreasonable to be able to download a firmware bundle and be able to update the printer just by placing it on the root of the SD card. See: most other electronics devices, motherboard BIOS, etc. Being able to push it from Studio (LAN mode or not) can/should be an easy feature add on top of that.

seonwoolee, 2024-08-17:
It is possible as of firmware version 01.08.02.00 (20240812) to update the firmware via SD card. Of course, you still need to do an OTA update to that version

[Release history](https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history)
[Wiki Guide](https://wiki.bambulab.com/en/x1/manual/X1-firmware-update-from-SD-card)

probonopd, 2024-08-28:
Is this also coming to the A1 and A1 mini?

Triskae, 2024-08-29:
> Is this also coming to the A1 and A1 mini?
> 
> Currently I don't see it mentioned (yet?) on https://wiki.bambulab.com/en/a1-mini/manual/a1-mini-firmware-release-history.

Yes, it will come to the A1 series 
Source: [https://wiki.bambulab.com/en/general/update-firmware-from-sd-card](https://wiki.bambulab.com/en/general/update-firmware-from-sd-card)

rspeed, 2025-04-18:
P1 firmware version 1.07.00.00 added offline firmware updates via the SD card.