bambulab/BambuStudio #4032: Bambu Printer disappeared in Bambu Studio after last update

MNCarlos, 2024-05-11:
### Bambu Studio Version

1.8.4.5.1

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

I want to choose my Bambu X1C but your program won't show it.

### How to reproduce

I opened Bambu Studio and loaded a model and the program didn't show my Bambu X1C. When I clicked on gear on the printer line to add the X1C, it doesn't even list any Bambu Printers. It lists a bunch of other brands, but your own printers have disappeared. 

### Actual results

Nothing! I can't print anything because you deleted the printers.

### Expected results

Would be able to pick a Bambu printer inside of Bambu's own product.

### Project file & Debug log uploads

This is NA. You have a software bug that deletes the ability to choose Bambu printers in yo
[image]
[image]
ur own program.

### Checklist of files to include

- [X] Log file
- [X] Project file

XunZhangBambu, 2024-05-11:
Can't reproduce in my machine. Please provide log or check if "BBL" exists in the folder under the installation path, for example "D:/bambu studio/resources/profiles/"

MNCarlos, 2024-05-13:
Yes, the BBL file is part of the program config files. It was located in the system folder and not in a resources folder. I don't have a resources file. Here is copy of the log from the old configuration files. I made a copy of everything before I deleting them. I was able to fix the problem by deleting the configuration folder and having the program rebuilt it. 
[log.zip](https://github.com/bambulab/BambuStudio/files/15296058/log.zip)

XunZhangBambu, 2024-05-14:
Perhaps it was a misoperation that deleted the corresponding configuration file. I'm glad you can solve your problem on your own!

closed 2024-06-17 by XunZhangBambu

Craig8944, 2024-12-16:
Same issue with my installation to latest version.  No Bambulab printers available to select in version 1.10.01.50.  After update, printer was there and worked fine but now it has magically disappeared, it shows a Creality Ender 3 as my printer option and no option to add any Bambulab printers.  Have uninstalled the software from Windows 10 PC to try a reinstall in hopes it will solve the issue.  Funnily, my printer shows and functions fine on the IOS App.  Quite annoyed with this.  So the printers are great but the software is like the original Ender 3! 😢🤣😂🤣

trakaismikus, 2024-12-19:
I'm also on 1.10.1.50 and same thing happened to me - complete reset, with a new wizard and I don't see any way to add my A1 mini. The guide under devices only show instructions on how to pair to octoprint and other solutions but not Bambu printers. I will try to reinstall but this is super annoying.

bosh, 2024-12-20:
I'm on 1.10.1.50 on Windows 11 and have also lost printer info as per OP screenshots. I logged out and logged back in and it did not restore settings either. Restarting the program after this also did not help. Going over to the printer, everything seems fine on that end, and the phone app can communicate and control it just fine.

I recall this also happened a few months ago and eventually just fixed itself. I'll try a reinstall tomorrow.