bambulab/BambuStudio #4037: Night Mode - All lights off

NasaEagle, 2024-05-11:
I have my X1C printer with AMS in the room I sleep in, and at times, the bright lights from the logo and the AMS bother me at times (I know the AMS lights aren't super bright, but when the room is pitch black it can be frustrating).

My alternative at the moment is by putting a piece of electrical tape over the logo, but the light still leaks over the top of the tool head.

It would be nice to add a setting to turn off all the lights (cabin, tool head logo, AMS). Another possible solution could be to turn off all the lights when the front control backlight turns off, and all the lights could turn on when the panel is touched or reactivated (this could be controlled by a button in the settings page). 

This would be great for those who have their printers in their room, or who wish to not have the lights running for the duration of a print.

Thanks!

zhimin-zeng-bambulab, 2024-05-15:
[image]
@NasaEagle  You can click this button on the device page to control the light switch. Does this meet your needs?

NasaEagle, 2024-05-15:
No, it’s more about being able to control the lights on the tool head and the AMS.

Carter Stengel

image.png (view on web)<https://github.com/bambulab/BambuStudio/assets/120697589/2967255e-c1c0-4267-a8f9-182fb3ed1cf2>

MrDB42, 2024-05-16:
New version of BBS should have added the following to the bottom of "Machine end G-code"
M960 S5 P0 ; turn off logo lamp

BUT... My issue is that sometimes the chamber light will turn on at the end of the print. I haven't figured out the pattern yet. Would _love_ to know the G-code to control the chamber light.

markwalkom, 2024-07-25:
I wanted to add a +1 to this because even when you turn off a printer, or it goes to sleep mode, the AMS lights *don't* turn off. And it's kinda annoying.

Frozander, 2025-03-06:
Same here. I wouldn't mind having 2 buttons, even under some submenus to turn off tool and AMS lights. (Especially the tool light)

picard102, 2025-08-03:
Yes, the tool head having to UI to control it seems wild.