bambulab/BambuStudio #4039: Full replicable app crash (close) while slicing with no warning

KodeMunkie, 2024-05-12:
### Bambu Studio Version

1.9.1.58

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Additional system information

CPU: 11th Gen i9 11980HK
GPU: 16GB Geforce RTX 3080 (laptop variant)
RAM: 32GB

### Printer

Bambu P1S

### How to reproduce

1. Load attached file (do NOT simplify the model).
2. Choose layer height 0.08 and initial layer height 0.2
3. Resize, untick uniform scale, select 50, 50 , 100 (unchanged) in the height percentages.
4. Click slice
5. A few seconds in the app will crash, occasionally a dialog is popped up briefly about an access violation before Bambu Studio closes.

### Actual results

In version 1.9.1.58 it simply quits without warning or flashes a dialog too quickly to screenshot.
In version 1.8.4.51 it performs an odd gcode calculation going extremely far out of the model.
[image]

At other scales >= 60% instead of 50% scales, it appears to slice correctly.

### Expected results

Model should slice correctly and be printable

### Project file & Debug log uploads

[plastipic-image - 2024-05-12T020715.285.zip](https://github.com/bambulab/BambuStudio/files/15285575/plastipic-image.-.2024-05-12T020715.285.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/15285627/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2024-05-13:
Hi，Could you update the dump file to help us target the issue

KodeMunkie, 2024-05-13:
Unfortunately I can't give you the memory dump due to privacy reasons, however the bug should be easily replicable without it. If it isn't, I would assume it's unique to my setup. I have a suspicion it's very large memory usage causing pointers to not be initialised correctly within valid bed coords (for 1.8) or read from (1.9, hence the access violation).

KodeMunkie, 2024-05-13:
I can confirm this bug still exists in 1.9.1.66

QingZhangBambu, 2024-05-15:
[image]
it crashed on detecting thin wall function. for now, you can close it to get right slice result.