bambulab/BambuStudio #4042: Windows平台切片多边形较多的模型时程序crash

liyesa, 2024-05-12:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 专业版

### Additional system information

CPU: 13th Gen Intel(R) Core(TM) i9-13900K   3.00 GHz
RAM: 64.0 GB 
Display card: Nvidia GeForce RTX4090

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1）选择模型 https://makerworld.com/zh/models/93385#profileId-99950 
2）选择包含齿轮的模型，下载3mf, Launcher Gears (small plate, single-color)
3）在bambu studio中打开，同步AMS耗材，选择切片单盘
4）程序崩溃退出，无任何提示
5）若用管理员模式，可能会导致蓝屏，机器重启

### Actual results

4）程序崩溃退出，无任何提示
5）若用管理员模式，可能会导致蓝屏，机器重启

### Expected results

能正常切片

### Project file & Debug log uploads

[Toroidal Launcher - Fredwood6000.zip](https://github.com/bambulab/BambuStudio/files/15287018/Toroidal.Launcher.-.Fredwood6000.zip)
[bambu-studio.exe.5200.zip](https://github.com/bambulab/BambuStudio/files/15287020/bambu-studio.exe.5200.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2024-05-13:
Please confirm the studio version and update the new dump file with that version. The dump file you provided does not match 1.8.4.51

QingZhangBambu, 2024-05-13:
and if you rotate the model, will the model be sliced successfully?is the old version or latest version works? and you may try this method to seem if it works. https://github.com/bambulab/BambuStudio/issues/3318

liyesa, 2024-05-13:
> and if you rotate the model, will the model be sliced successfully?is the old version or latest version works? and you may try this method to seem if it works. #3318

Thanks!  It works , https://www.reddit.com/r/overclocking/comments/1axepvu/optimizing_stability_for_intel_13900k_and_14900k/

By changed BIOS setting for Asus Z790,  the model can be sliced successfully.

QingZhangBambu, 2024-05-14:
nice to hear that!

closed 2024-05-14 by QingZhangBambu