bambulab/BambuStudio #4043: Text shape tool initial shows wrong font

EriHau, 2024-05-12:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Ubuntu 22.04.4 LTS

### Printer

X1C

### How to reproduce

1. Select object
2. Click on Text Shape tool
3. Add text
The font "Abyssinica SIL" is shown as selected font (see first screenshot)
[image]
4. Select the same font as already shown again
5. The font on the object changes to the "real" Abyssinica SIL (see second screenshot)
[image]

### Actual results

After step 3 the wrong font is shown on the object

### Expected results

when opening the Text shape tool the font shown in the popup should be the font which is being used at the object. It should not be necessary to select the font again (step 4 above). 

### Project file & Debug log uploads

[Pflanzenbeschriftung.zip](https://github.com/bambulab/BambuStudio/files/15287407/Pflanzenbeschriftung.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2024-06-03:
Can you reproduce this issue in version 1.9 ?

lasseruud, 2024-06-16:
I'm running 1.9.2.57 on Windows 10 and I have somewhat the same problem. Bambu Studio displays and uses the wrong font. I'm unable to place text with the font I want to use:

I'm trying to use a font called Belgium and this is how Bambu displays it:
[image]

This is how the font is supposed to look like:
[image]

MackBambu, 2024-06-16:
@lasseruud yes,it's another known issue. 
https://github.com/bambulab/BambuStudio/issues/818

lasseruud, 2024-06-17:
Thank you @MackBambu for pointing me to the issue. The font I was trying to use, was installed for my user only. The font was installed in C:\Users\%username%\AppData\Local\Microsoft\Windows\Fonts. When I installed the font to all users (right click and choose "Install for all users"), I was able to use the font in Bambu Studio. Might work for OP too?

MackBambu, 2024-06-17:
Yes, there are some problems with finding fonts, some fonts can't be found and some fonts can't be used even though they are found.