bambulab/BambuStudio #4050: crashes on open

jascha-sacc, 2024-05-13:
### Bambu Studio Version

1.09.01.66

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS Sonoma 14.3

### Additional system information

Macbook Air, M2, 2022

### Printer

X1 Carbon

### How to reproduce

I open the app, it logs in, crashes and quits out.

### Actual results

It crashes.

### Expected results

It opens.

### Project file & Debug log uploads

I don't have one.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ToxicPaulonium, 2024-05-13:
same problem 
Sonoma 14.2.1

FlorianWicke, 2024-05-13:
Same Problem.
Apple m1
Sonoma 14.4.1

KolemanIreland, 2024-05-13:
Same Problem
Apple M1
Ventura 13.3.1 (a)

taurus1666, 2024-05-13:
+1 crash always, it’s not possible to print anything. 

Anyone who has an old version for macOS?

sashcardi, 2024-05-13:
+1 
Crashes on open. M2 Mac Mini, Sonoma 14.4.1

jhef807, 2024-05-13:
apple m1 some problem

matt123458, 2024-05-13:
Same, Sonoma 14.3 - M2 Max 64GB

nbowha001, 2024-05-13:
Same, Ventura 13.6.6 2017 MacBook Pro.

BastR, 2024-05-13:
Same, macOS Sonoma 14.4.1 M1 max

cfgraziano, 2024-05-13:
Same. Sonoma 14.4.1 (23E224) M3

jeluchu, 2024-05-13:
Same problem
With macOS Sonoma (14.4.1) Apple M1 Pro

t-unix, 2024-05-13:
BambuStudio 1.09.01.66 stopped crashing after wiping its settings folder.
Be aware that this will reset your local BambuStudio configuration. You will need to login freshly. Settings can be imported if synced to Cloud.
Run following command to move the BambuStudio settings folder to $HOME/Library/Application\ Support/BambuStudio_backup
```mv $HOME/Library/Application\ Support/BambuStudio/ $HOME/Library/Application\ Support/BambuStudio_backup```
- Start BambuStudio 1.09.01.66 and follow the setup wizzard...

louisdedecker-eptura, 2024-05-13:
> BambuStudio 1.09.01.66 stopped crashing after wiping its settings folder. Be aware that this will reset your local BambuStudio configuration. You will need to login freshly. Settings can be re-imported if synced to Cloud. Run following command to move the BambuStudio settings folder to $HOME/Library/Application\ Support/BambuStudio_backup `mv $HOME/Library/Application\ Support/BambuStudio/ $HOME/Library/Application\ Support/BambuStudio_backup` Then start BambuStudio 1.09.01.66 and follow the setup wizzard...

thanks, this solved my crashes too.

lanewei120, 2024-05-14:
Hello @jascha-sacc @zviratko @ToxicPaulonium ...
Could you help to give an crash stack of this issue?
you can follow the guide here https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

we will fix this issue as soon as possible

lanewei120, 2024-05-14:
and for anyone meet crash
please try to remove $HOME/Library/Application\ Support/BambuStudio/ota before run bambustudio
it is likely caused by the file with wrong format cached in this directory

Schokobecher, 2024-05-14:
@lanewei120 That solution worked for me 
Bambu Studio v01.09.01.66 on macOS 14.4.1

lanewei120, 2024-05-14:
@Schokobecher
thanks

which version are you using before?
1.8.4.51 or 1.9 public beta?

Schokobecher, 2024-05-14:
I migrated from 1.8.4.51 to 1.9.1.58 (Beta) a few days ago, and then to the Official Release of 1.9 today

lanewei120, 2024-05-14:
Hi,

We have fixed this issue also, the new image address 
https://github.com/bambulab/BambuStudio/releases/download/v01.09.01.66/Bambu_Studio_mac-v01.09.01.67-20240514131542.dmg

please update to this image

Sorry for the inconvenience caused to everyone

closed 2024-05-14 by lanewei120

bmmmm, 2024-05-14:
sure? to me it sounds like you fixed the workaround... ;) but it just happened to me with 01.09.01.67, with macOS 14.5. I just had to restart Bambu Studio.

gabfv, 2024-05-14:
I had the same crash on startup (well, when the connected printer becomes online) with 1.09.01.66 and with Win11 23H2. Deleting the ota folder in %appdata% worked to fix the startup crash.

lanewei120, 2024-05-15:
> sure? to me it sounds like you fixed the workaround... ;) but it just happened to me with 01.09.01.67, with macOS 14.5. I just had to restart Bambu Studio.

@bmmmm do you mean your bambustudio still crash when using 1.9.1.67?
if using 1.9.1.67
no need to do the above steps(workaround)

bmmmm, 2024-05-15:
@lanewei120 correct, I just updated to 1.9.1.67  yesterday, after the update has been installed, I experienced this behavior. I just needed to restart BambuStudio, then it started without any errors