bambulab/BambuStudio #4052: Allow more than 6 printers in Multi-Device mode

Underwater-Hockey, 2024-05-13:
**Is your feature request related to a problem? Please describe.**
Not a problem with the implementation, just a request for an increase in allowed printers.

**Describe the solution you'd like**
I would like the limit on the printers allowed to be removed or increased to support those of us with larger farms.

**Describe alternatives you've considered**
N/A

Giulio-Machiavelli, 2024-05-14:
Would be an important function

FyrbyAdditive, 2024-05-31:
I came here to upvote this on behalf of my manufacturing partner. While 6 printers is an ample amount for most hobby users the ability to manage more in the context of a farm would be useful.

robatjamf, 2024-06-02:
I came here to upvote this because well the printers are awesome, anyone in their right mind would buy a crap load of P1S machines and just use them vs. others... so win win for all print farms, just make the tools to kick butt

motionbug, 2024-06-02:
I also came here because my friend Tim said that print farms need to have the best options, one is good printers (bambu printers) and second is to be able to control the farm! So make sure farms can be farms!

JonasKuschel, 2024-07-05:
Please make this available. We are past 6 printers🙏🏻

rtrocc, 2024-07-27:
Put me down as a VERY EAGER recipient of this feature as well. I have more than 6 and don't understand why there is a limit to begin with.

And it would be fabulous if there was a way to send multiple plate jobs to the printers SD cards using multidevice button, right now I can only have the printer start a single sliced plate and it doesn't even let me save it to the sd card.

replaysMike, 2024-10-25:
Indeed - the 6 device limitation really hampers usage for building a print farm. I love that they have started to integrate this into Bambu Studio but it should be unlimited (or at least 50). If you need someone to help tackle this, just ping me.

TransparentHouseSF, 2024-12-19:
+1. Very strange and limited. We need more them six, at least 20 or even unlimited.

upcyclefab, 2025-01-15:
I now have 6 printers and (2) more due to arrive later this week.  This makes ZERO Sense why this hasn't been solved yet.  Charge me for the Print Farm "Server/PRO" version of Bambu Studio, I will gladly pay for it.  With so many end of year sales..they should want to increase the limit of 6 printers!

GizmoDevs, 2025-01-24:
With my small farm increasing to 8 and myself only just finding this feature (I've been wondering why it hasn't been available for the last 12 months and it's been right under my nose for most of it!), I'm about to fall outside of the scope of the multi-device allowance, which is going to bring a pain point back to the printing process. I want to get away from having to click the same buttons multiple times, not go back to that same process because the software limits me. Stick a hard 5 minute delay between each printer to cancel out the possibility of a power surge, I'm cool with that, I just want to be able to use the software with the potential ease of use that it has.

PLAmenPRINT, 2025-02-01:
One more vote for 6+ printers in "multi devices" tab.

bolsoncerrado, 2025-03-18:
One more upvote. I could understand some sort of "cloud" connection must be stablished for Bambu to limit this per customer, but heck! DISABLE THE LIMIT FOR LAN MODE at least!!!!