bambulab/BambuStudio #4054: Chamber fan goes to 100% when warming bed at begining of print job

swheettaos, 2024-05-13:
### Bambu Studio Version

1.91.66

### Where is the application from?

Bambu Lab github releases

### OS version

Mac OS

### Additional system information

MacBook Pro running latest version of MacOS

### Printer

Bambu P1S

### How to reproduce

Any job with any default profile, slice, and print.  Chamber fan goes to 100% even if the filament is set to different amount or complete disabled. The chamber

### Actual results

Fan goes to 100% until job starts to actually print.  Bed level and filament load the chamber fan is @ 100%

### Expected results

Chamber fan so not go higher than was is specified in filament profile.

### Project file & Debug log uploads

[MenMagV1LoadingTool.3mf.zip](https://github.com/bambulab/BambuStudio/files/15298058/MenMagV1LoadingTool.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

TryGreeen, 2024-05-13:
i have the same problem

avelytchko, 2024-05-13:
Same problem after upgrading to 1.9.1.66. Advanced tab in filament settings doesn't work anymore. M106 P3 SXX gcode does nothing.

swheettaos, 2024-05-13:
Do you guys all have p1s or other models?

avelytchko, 2024-05-13:
> Do you guys all have p1s or other models?

Bambu Lab P1S with AMS.

XunZhangBambu, 2024-05-14:
@swheettaos 
This is caused by this PR #3948 . According to original logic, the current fan speed is the expected speed to prevent PLA jamming. In previous versions, the logic for handling the hot bed temperature greater than 50 when printing PLA was not operated.

FenmanUK, 2024-05-14:
Looks like it's intentional, there was a bug in the machine start gcode (stored against the nozzle) which they've fixed. There's two blocks that set the chamber fan. 

In the old version, the block read as follows:
```
;=============turn on fans to prevent PLA jamming=================
{if filament_type[initial_extruder]=="PLA"}
    {if (bed_temperature[initial_extruder] >45)||(bed_temperature_initial_layer[initial_extruder] >45)}
    M106 P3 S180
    {elsif (bed_temperature[initial_extruder] >50)||(bed_temperature_initial_layer[initial_extruder] >50)}
    M106 P3 S255
    {endif};Prevent PLA from jamming
{endif}
```

If the bed temperature was over 45 degrees, the fan would always be set to S180 (70%?) as anything over 50 is also over 45 and therefore falls into the first clause.

In the new version, they've fixed the logic:
```
;=============turn on fans to prevent PLA jamming=================
{if filament_type[initial_extruder]=="PLA"}
    {if (bed_temperature[initial_extruder] >50)||(bed_temperature_initial_layer[initial_extruder] >50)}
    M106 P3 S255
    {elsif (bed_temperature[initial_extruder] >45)||(bed_temperature_initial_layer[initial_extruder] >45)}
    M106 P3 S180
    {endif};Prevent PLA from jamming
{endif}
```

So now, a bed temperature over 50 degrees will set the fan to S255 (100%) and anything over 45 and up to 50 degrees will set the fan to S180. 

As the default temperatures for PLA tend to be 55 degrees, the fan will run at 100% until the print starts and the filament values override it.

swheettaos, 2024-05-14:
ok.. Never had it jam for me on pla using old settings .. I only have about 350 hours on mine so far.  Was this a big problem for most people?

XunZhangBambu, 2024-05-14:
Some users have reported experiencing jamming when printing with PLA.  Therefore, at higher temperatures, we have set a higher fan speed to accelerate cooling and avoid jamming.

Nofreakinway, 2024-05-14:
Please Make it Stop. 70% was fine. How can we turn it back. Please let us be able to turn it back with out having to manually do it for every print.

Please & Thank you

swheettaos, 2024-05-14:
Look like we can just modify the gcode on the printer settings

Stephen

Nofreakinway, 2024-05-14:
We talking like here? i may need a little help with this part if thats cool. Just point me at the bit to change haha :) 
[image]

nicoodeimos, 2024-05-14:
Can confirm this also. So annoying, never had any issues with the previous behavior. So loud 😿

OzDruiD, 2024-05-14:
> We talking like here? i may need a little help with this part if thats cool. Just point me at the bit to change haha

So - I wont speccifically give you code to copy and paste, because you should understand the code and what its doing, so if you make the change your accceptign that risk (if that makes sense) - but I guess others might :P 

In your screenshot the values  M106 P3 Sxxxx represent the chamber fan speeds in this case.   
the Sxxxx part is the speed indicator,  255 being 100% fan speed,  155, being a signifcant bit less than that... work out the math.
Google various Sxxx values and work it out what percentage you want.

IF you read the logic therefore  the bit about >50 - is heat bed, above 50 degress turns on Chhamber fan to 100% (255).  As confirmed by bambu rep.    

So .. if you want it LESS than 100% chamber fan, even with heatbed ABOVE 50 - potentially against what Bambu has done to prevent PLA issues (for some users)  there are MULTIPLE Places you can change these fan values, so you really want to be careful and modify the one for the print start job only, if thats the only one thats annoying you....

cayalag, 2024-05-14:
This is extremely annoying. 

Could you please make an option to disable it? Maybe under Develop mode ?... PLEASE !

y3dltd, 2024-05-14:
Quick fix (but irritating few hours of my print farm sounding 2x as loud till I learned it wasn't a bug after all, a heads up would of been nice!) 

So my simple fix (use at your own risk) re PLA in Machine Start Gcode from M106 P3 S255 to M106 P3 S180 (setting it back to 70%)

Not sure why Bambu hasn't suggested officially as a workaround.

Happy Printing!

capsel22, 2024-05-14:
those fans are not the best quality to begin with. setting them on 100% is absolute nuts, the high pitch noise does my head in.
I already adjusted it down from 255.

capsel22, 2024-05-14:
Looks like Bambu listened. I just saw a small update inside BL reducing 100% to 70% 

[image]

OzDruiD, 2024-05-14:
YUp code has changed again, no condition for 50 degress+   for 100% fan anymore.
Everything above 45 degrees get 70% fan.

DanTup, 2024-05-14:
Where are you seeing these changes? I don't see any updates to the files at https://github.com/bambulab/BambuStudio/commits/master/resources/profiles/BBL/machine 🤔

J108560, 2024-05-14:
And the next question is: What is M142 P1? This machine code is also in every PLA profile.

DanTup, 2024-05-14:
I got a BambuStudio update to 1.9.1.67, which then prompted me for a "configuration update" which indeed removed the 100% fan part of the start gcode.

I guess the change (and releases) are handled internally and haven't been pushed to GH yet.

swheettaos, 2024-05-14:
it looks like in releases there is a 1.9.1.67 does include settings update that fixes this issue.  I notice the rest of the O/S's still look like they are .66 .. so I don't know if this was a Mac only issue.

Nofreakinway, 2024-05-15:
So i guessing there will be an Update that will be prompted at some point? Or do i need to Fine the latest one on github and re download this?

DanTup, 2024-05-15:
> So i guessing there will be an Update that will be prompted at some point? Or do i need to Find the latest one on Github and re download this?

You will be prompted. I was prompted to update to .67 and then later given a "configuration update" prompt. I don't know if the .67 update was required to get the configuration prompt, it just showed up ranodmly when I was opening a model (but after I've already started a print using the new version).

OzDruiD, 2024-05-16:
V19.1.**66**  appeas to the Windows version, **.67** the Mac
I got this "config update" automatically on both,. yet both retrain their version number respectively. - its just a quick config update to the machine code for fan control.
[image]

> And the next question is: What is M142 P1 R35? This machine code is also in every PLA profile.

Not sure, P might be a internal bambu thing.  
Ref here # https://reprap.org/wiki/G-code#M142:_Firmware_dependent
nothing here, https://forum.bambulab.com/t/bambu-lab-x1-specific-g-code/666

pmoliterni, 2024-05-18:
@OzDruiD How dit you get this message ? Nothing happened in my case.

NVNDO, 2024-05-25:
Same problem here!

Just made a calibration and suddenly the fan cranked up to 100% (which had never happened before)
I could not switch it off through the App, as the App said, printer Offline (which wasn't the case) 
Then I tried again after rebooting the App and although I could now access the printers settings, no function was showing. 
Restarted the App again and now I could change the Fan. Set it to 20%.

5 minutes later it again shot up to 100%

Btw: I only print with PETG and therefore only use PETG profiles.

pmoliterni, 2024-05-25:
@NVNDO Have a look at the right bottom corner in Bambu Studio, it shluld appear a changing in configuration file. This will solve the problem.

[image]

NVNDO, 2024-05-25:
I have the slight feeling I might have over-read that (as I did not read the entire thread) 

Thx a lot for pointing this out again, I had been ignoring these pop-ups as mostly they're just informative in regards of things I already know.. but didn't spot the ability to update. 

🙏🏼

closed 2024-06-26 by XunZhangBambu

mysterious-roman-dodecahedron, 2024-08-05:
> And the next question is: What is M142 P1 R35? This machine code is also in every PLA profile.

Hi, I found out that M142 is a firmware dependent code for "Set Cooler Temperature (Fast)". 

This sets a target temperature for the heatbreak where R### is the standby temperature (presumably fan off) and S### is the target temperature (above which fan powers on). This code can be associated with a T### parameter to specify the toolhead, which is absent here. The S### parameter here appears to be specifying which fan the system should power.

This sequence is triggered when PLA is loaded to keep the filament on the cold side below the glass transition temperature (stop heat creep) to prevent a clog from potentially occurring.

mysterious-roman-dodecahedron, 2024-08-05:
> > And the next question is: What is M142 P1 R35? This machine code is also in every PLA profile.
> 
> Not sure, P might be a internal bambu thing. Ref here # https://reprap.org/wiki/G-code#M142:_Firmware_dependent nothing here, https://forum.bambulab.com/t/bambu-lab-x1-specific-g-code/666

In this case, it appears that the P### parameter is "Fan index" specifying which fan to power. 
I think P1 might be the part cooling fan. They annotate P2 as "big fan ,to cool down toolhead", and P3 is also called at 70% power if PLA is loaded and bed temp is currently measuring or will be set at first layer to 45 degrees or more.

```
;=============turn on fans to prevent PLA jamming=================
{if filament_type[initial_no_support_extruder]=="PLA"}
    {if (bed_temperature[initial_no_support_extruder] >45)||(bed_temperature_initial_layer[initial_no_support_extruder] >45)}
    M106 P3 S180
    {endif};Prevent PLA from jamming
    M142 P1 R35 S40
{endif}
M106 P2 S100 ; turn on big fan ,to cool down toolhead
```

pleasantone, 2024-12-04:
@mysterious-roman-dodecahedron did you ever confirm M142 P1 R35 S40 turns on the heatbrake fan to 40 at 35c?

Allexkramer432, 2025-01-01:
@pleasantone M142 means firmware dependent, which means there is a code somewhere in firmware that links up to this if applicable. P1 is AUX fan. R35 means _get the chamber to 35 degrees_ S40 says to speed the P1/AUX to that speed AT LEAST if not already, in conjunction with the P3 (from above code), to achieve this.