bambulab/BambuStudio #4055: 3DConnexion dialog blocked by MainFrame eating "Cmd-Shift-M" improperly

SuperTango, 2024-05-13:
### Bambu Studio Version

1.9.1.66 (and everything before)

### Where is the application from?

Bambu Lab Official website, Built from source by yourself or others

### OS version

macOS Ventura 13.3.1

### Additional system information

2019 MacBookPro 64GB RAM, 4TB SSD (2.4 GHz 8-Core Intel Core i9, AMD Radeon Pro 5500M 8 GB, Intel UHD Graphics 630 1536 MB)

### Printer

Bambu Lab  P1S

### How to reproduce

1. Open Bambu Studio and go to the "Prepare" or "Preview" tab
2. hit "Cmd-Shift-M"

### Actual results

The Bambu Lab screen minimizes into the MacOS Dock

### Expected results

The BambuLab "3DConnexion " dialog should appear.

This happens for all projects (or even when no project is loaded).

### Project file & Debug log uploads

No project file is needed.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SuperTango, 2024-05-13:
The problem is that hitting "Cmd-Shift-M" on mac always minimizes the app, even on the "Prepare" and "Preview" screen (where Cmd-Shift-M should bring up the 3Dconnexion preferences dialog.)x

I tried all of the mac system based ways to get Bambu Studio to ignore "Cmd-Shift-M" like the suggestions here: https://apple.stackexchange.com/questions/115562/how-do-i-disable-the-minimize-command-m-shortcut, but none of them worked because Bambu Studio manages all of the keyboard shortcuts on its own so the OS is not involved with handling the "minimize" keyboard shortcut (as far as i can tell).

I'm providing a pull request that fixes the problem:

The code that minimizes (using the WXWidgets "Iconize()" call) happens in MainFrame.cpp keyboard event loop.  The code that's checking, looks for "Cmd-M" but does not check for any other keyboard modifiers, so I added a check to ignore the event if Shift is pressed along with "Cmd-M".  

There's a secondary issue that isn't really relevant to this bug in that the app will still minimize when pressing "Cmd-Shift-M", but ONLY on the "Home" sub-screen. (all other sub-screens work as they should).

I'm not sure why, but when the "Home" sub-screen is selected, the keyboard event loop (MainFrame.cpp, line 609), is called TWICE when "Cmd-Shift-<any key>" is pressed:

* Once where the event's wxKeyModifier (retrieved via `evt.GetModifiers()` is set to `wxMOD_CONTROL`  AND `wxMOD_SHIFT`.  (this is correct)
* Once where the event's wxKeyModifier  is **ONLY** set to `wxMOD_CONTROL` (this is wrong).

Again, this double-event (with the wrong modifiers) only happens when the user is on the "Home" sub-screen.  For the context of this bug the 3DConnexion preferences dialog isn't needed on the "Home" sub-screen so this secondary bug doesn't matter.  But it does make the UX odd where Cmd-Shift-M will minimize the app when the user is viewing the "Home" sub-screen, but not minimize the app when the user is viewing any other sub-screen.

SuperTango, 2024-05-13:
Pull request #4058  created.

LiZ-Li-BBL, 2024-06-13:
merged commit 932be8

closed 2024-06-13 by LiZ-Li-BBL