bambulab/BambuStudio #4056: Erroneous paths in vase mode

dibunker, 2024-05-13:
### Bambu Studio Version

1.9.1

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: AMD Ryzen 9 3950X
Memory: 128GB
GPU: EVGA GeForce RTX 2080 Super

### Printer

Bambu Lab X1C

### How to reproduce

1. Go to https://makerworld.com/en/models/460194#profileId-368625
2. Open the "with base" print profile in Bambu Studio v1.9.1 Official Release
3. Slice plate 3 (

### Actual results

On several layers, there are erroneous G1 moves

For example, on layer 49 of plate 3:
[image]

### Expected results

Correct/Expected path of layer 49 of plate 3 in v1.8.4.51 of Bambu Studio:

[image]

Orca Slicer v2.0.0 also generates the correct path:

[image]

### Project file & Debug log uploads

[sierpinski_pyramid_with_base.zip](https://github.com/bambulab/BambuStudio/files/15299144/sierpinski_pyramid_with_base.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/15299167/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2024-05-14:
[image]
discontinuous wall cause that, you can close slow down overhang speed to get continuous wall. 
[image]

QingZhangBambu, 2024-05-14:
or you can rotate a bit the model, the result wull be right.

QingZhangBambu, 2024-10-15:
it has been fixed and will release in the next version. please wait.

ttt888zzz, 2024-10-16:
The latest version has been fixed.
 Please visit the website  at https://github.com/bambulab/BambuStudio/releases and try again.

closed 2024-10-16 by ttt888zzz