bambulab/BambuStudio #4057: Bambu Studio corrupts models when printing from microSD, but Bambu Handy works

RyanEwen, 2024-05-13:
### Bambu Studio Version

1.9.1

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

P1S

### How to reproduce

Send plate to printer from Bambu Studio.
Print plate from microSD card using Bambu Studio and/or Bambu Handy.
Not sure when exactly the issue arises, but eventually the plate file will become unprintable from Bambu Studio. 

### Actual results

After starting the print, the printer gives an error about being unable to get the AMS mapping (after heating, but before bed levelling). It seems like the latest 1.9.1 release (non-beta) Bambu Studio gives an error parsing the file before starting the print.

Printing from Bambu Handy will still work, however. 

### Expected results

Plate file should remain printable from Bambu Studio as it does from Bambu Handy.

### Project file & Debug log uploads

I will try to gather logs and share a plate. For now, I have only been printing models which I cannot share. Hopefully others have run into the same issue and can help diagnose.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2024-05-14:
@RyanEwen 
I tried to reproduce the problem in the way you said, but the printing was successful. Can you provide the print file and log file.

walterwongbbl, 2024-05-14:
Hi. @RyanEwen 
Can you provide a screenshot of the error.

RyanEwen, 2024-05-14:
I will update with screenshot & logs next time this happens. I have re-sent the problem plates so it will take some time before they stop working again

walterwongbbl, 2024-05-15:
@RyanEwen 
Thank you very much, waiting for your upload.

closed 2024-08-08 by walterwongbbl