bambulab/BambuStudio #4060: Privacy Policy missing navigation back button

cndjonno, 2024-05-14:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

N/A

### How to reproduce

1. Click on Login
2. Click on Privacy Policy
3. ...
4. Stuck on Privacy Policy

Had to resort to keyboard shortcuts to guess a back command.

### Actual results

Shows Privacy Policy

### Expected results

Expect button to click back

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-05-14:
could you give a snapshot?
and is this issue appeared before?

thanks

ttt888zzz, 2024-05-14:
Thank you for your very detailed advice.  
you can press "Esc" to close most dialog in Bambu Studio