bambulab/BambuStudio #4064: failed boss placement

twilkers1976, 2024-05-14:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

win 11

### Printer

X1C

### How to reproduce

attempt to relocate boss in z dimension failed.  it just jumps back to base plate.  see attached 3mf file

### Actual results

boss jumps back to baseplate instead of staying in desired locatoin.

### Expected results

expect boss to stay in z dimension where it is released.

### Project file & Debug log uploads

[2024_5_14Wilkerson boss placement failure.pdf](https://github.com/bambulab/BambuStudio/files/15305749/2024_5_14Wilkerson.boss.placement.failure.pdf)
[2024_5_14_wilkerson_Inner_Left 140 percent with boss.zip](https://github.com/bambulab/BambuStudio/files/15305764/2024_5_14_wilkerson_Inner_Left.140.percent.with.boss.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2024-05-16:
Hello, could you record a video to explain your issue? Thank you

Haidiye00, 2024-05-20:
Do you still need to solve it, or have you already solved it yourself? If not resolved, please record a GIF or video. Thank you

twilkers1976, 2024-05-20:
ResolvedSent from Tim's iPhone

closed 2024-05-20 by Haidiye00