bambulab/BambuStudio #4065: Can't export as STL with latest BambuStudio 1.9.x

bolsoncerrado, 2024-05-14:
### Bambu Studio Version

1.9.1

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sonoma 14.4.1

### Additional system information

M1Ultra

### Printer

P1P

### How to reproduce

1. Open .3MF
2. Select MORE THAN ONE .STL
3. Right click -> Export as one STL
4. Give a name and save.

### Actual results

Nothing is saved.

### Expected results

A .STL containing the selected STLs saved as ONE STL.

(Works fine if you just select ONE stl)

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-05-14:
Sorry, the issue has been found. You can use the right-click “merge” operation to export  one STL, import multiple objects into an STL, it will be fixed in the next version

Haidiye00, 2024-05-14:
[image]
Also, entering through this entrance, this function is normal.

closed 2024-10-16 by Haidiye00