bambulab/BambuStudio #4066: Tree support brim width not working

alvarezcabrera, 2024-05-14:
### Bambu Studio Version

win-v01.09.01.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 home

### Printer

Bambu lab X1 carbon

### How to reproduce

1. start a part with "tree(auto)" supports and use some painted supports
2. Set "Tree support brim width" to a large value (e.g. 30 mm)
3. Slice part. Note generated tree supports around the base
[image]

### Actual results

Tree brim remains to a minimum value

### Expected results

Tree brim changes to the specified value. Same file sliced in win-v01.08.04.51 does not have any issues

### Project file & Debug log uploads

[TreeSupportBrimMissing.zip](https://github.com/bambulab/BambuStudio/files/15307928/TreeSupportBrimMissing.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2024-05-17:
Bug confirmed. Will be fixed soon. For now you can use this option with tree slim style.

Duudeeee, 2024-08-26:
> Bug confirmed. Will be fixed soon. For now you can use this option with tree slim style.

Not fixes.
Options vanish on new update

ttt888zzz, 2024-08-27:
Please try updating the latest version 1.9.4
https://bambulab.com/en/download/studio

closed 2024-08-28 by ttt888zzz

Rupp3r, 2024-10-17:
Tree support brim width gone in 1.9.7.52 :/
[image]

Adam-42, 2024-12-29:
From looking at https://github.com/bambulab/BambuStudio/commit/4bebe9eb6ae4bb2938980e89c30ab9d5afc34d6b I think the setting you now need to change to get the same effect as "Tree Support Brim Width" is "raft_first_layer_expansion" which is apparently called "Initial layer expansion" in the user interface: https://github.com/bambulab/BambuStudio/blob/4bebe9eb6ae4bb2938980e89c30ab9d5afc34d6b/src/libslic3r/PrintConfig.cpp#L2649C21-L2649C44

I've not verified that in any way - I just looked at the code and the commit message for the change that removed it.