bambulab/BambuStudio #4070: Bambu Studio Very Slow

Canadian-Gamer, 2024-05-14:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

I dont have that info at hand. where do i find it

### Printer

P1S

### How to reproduce

Slice a hueforge file. 

### Actual results

Very slow, errors about floating regions, this all started last night after the update 

### Expected results

Faster slicing. upload to makerworld successfull

### Project file & Debug log uploads

[Dallas Cowboys NFL 3MF.zip](https://github.com/bambulab/BambuStudio/files/15314501/Dallas.Cowboys.NFL.3MF.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-05-15:
can you slice it successfully on PC?

Canadian-Gamer, 2024-05-15:
Yes I can.  I sliced it and printed it. I can send you a picture of the printed copy.

QingZhangBambu, 2024-05-15:
so just upload to MW failed?

QingZhangBambu, 2024-05-15:
Does it cost hours to slice it on PC?

Canadian-Gamer, 2024-05-15:
I don’t know what you mean by cost hours.

It takes longer than it did before the update to Bambu studio yesterday.  It seems like all my Hueforges take longer to slice since the update, and I had been uploading one a day at least to Makerworld for about 3 months now

Canadian-Gamer, 2024-05-15:
Yes in this instance just the upload to makerworld is failing.  It is saying the slicing time exceeds the limit

QingZhangBambu, 2024-05-15:
OK, I will talk to the concerned staff about that.

QingZhangBambu, 2024-05-15:
they said that it did have problems last night,  but already fixed it now. Could you try it again?

Canadian-Gamer, 2024-05-15:
[image]

Yah it still won't upload, it is timing out, this isn't my longest or most detailed design, as a matter of fact my printing history shows that it only took 7.3 hours so I'm surprised this one is giving me the issue

QingZhangBambu, 2024-06-20:
@Canadian-Gamer Hi, it can upload to MW now, the bug had been fixed. I've confirmed that.

closed 2024-06-20 by QingZhangBambu