bambulab/BambuStudio #4071: After this update user preset process's disappear

bkimbler, 2024-05-15:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Ender 5 plus

### How to reproduce

open bambu studio 
select create new project
select user defined printer
select user defined filament
select process built in (system process)
make change to process
click save
enter ne name and select user preset
slice model  * do or dont do doesnt change anything*
export gcode * do or dont do doesnt change anything*
close bambu studiows

next reopen user preset processes and some user preset filaments missing
unable to export user presets for printers other then bambu labs

### Actual results

This is when i make a user preset named bobby
[image]

close bambu studios reopen and bobby is gone sometimes random presets disappera

[image]

### Expected results

That user preets once saved would still be there next time i open bambu studios for any printer they have in there to add

### Project file & Debug log uploads

[debug_Tue_May_14_20_01_32_10344.log.zip](https://github.com/bambulab/BambuStudio/files/15314767/debug_Tue_May_14_20_01_32_10344.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

DanBao-Bambu, 2024-05-15:
When I checked the log you sent, I found that the cloud version number reached 2.0, and the preset version number stored in the cloud did not match the local Slicer version number, resulting in the inability to load. May I ask if other Slicers were used to create user presets?

bkimbler, 2024-05-15:
No it was Bambi studio. All my presets disappeared after update. The images above is me making the presets. When I shut it down and reopen Bobby disappeared. If I close and reopen again others disappear. Only way I can keep them from disappearing is not closing the Bambi studio. The other issue is in device tab I can see what filament is loaded. In the main tab when I sync the filaments or sync it tells me there's no compatible filaments and won't sync unless I close and reopen Bambi studios but then I loose all my presets.

zhangrui1114, 2024-05-16:
Not only losing processes presets, but also machine&filament presets. That's a very very stupid update to delete local preset files!!!!
I lost all my presets after updating. Why not keeping local profiles like prusa/superslicer did？

DanBao-Bambu, 2024-05-20:
@bkimbler @zhangrui1114 I'm sorry about this. I've been analyzing this issue for the past few days, but I still lack some information. Can you create a log using the following rules and send it to me?
I need two log files to record two operations. The first one is which user preset you created, and after creating it, close Studio. The second one is to open Studio again (at this point, the preset should no longer be displayed), and then close Studio.
If you can send me these two logs, I will be able to track the creation and disappearance of the logs, and have more confidence in solving this problem. Thank you.
My email is maosheng.wei@bambulab.com.

DanBao-Bambu, 2024-05-31:
Does the problem still exist? If this situation still exists, please send me the logs when creating and disappearing. Thank you very much.

zhangrui1114, 2024-06-03:
> Does the problem still exist? If this situation still exists, please send me the logs when creating and disappearing. Thank you very much.↳

Thanks for ur reply, Sorry, I rollback to 1.9.0 because I don't want to rebuilding all presets. Maybe I will try again after version is stable, maybe 2.0.
And could you please tell me where is the logs?

DanBao-Bambu, 2024-06-03:
@zhangrui1114 You can refer to this wiki for log export. https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log