bambulab/BambuStudio #4075: retraction when switching material section missing

xkronnikx, 2024-05-15:
### Bambu Studio Version

1.91.01.66

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

windows 10

### Additional system information

12 Gen Intel Core i5-12400f
32gb ddr4 RAM
GTX1080

### Printer

Bambu lab P1S

### How to reproduce

1. Go to printer settings 2. Go to Extruder tab 3. Scroll down to realize that the newly implemented 'Retraction when switching material' section is missing.

### Actual results

The newly implemented 'Retraction when switching material' section is missing.
[image]

### Expected results

According to release notes on the Bambu wiki, there should be a 'Retraction when switching material' section. This however is missing. There is no way to easily implement the new retract before cut feature, other than manually entering the commands on the machine Gcode section.

I orignally updated Bambu studio from the Bambu lab website. When i realized the new section is absent i reinstalled it via the github releases. Contradictory to the Bambu wiki release notes, either seem to have this section implemented yet.

### Project file & Debug log uploads

Not applicable here.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SaltWei, 2024-05-15:
[image]

SaltWei, 2024-05-15:
Some filament is not suitable to use this function which is easy to cause nozzle clog.
Please enable in per filament setting.

closed 2024-05-15 by SaltWei

xkronnikx, 2024-05-15:
Oh i feel foolish now. I didn't realize i had to have developer mode on. Thank you.