bambulab/BambuStudio #4076: BambuStudio forgets Printer's password when disconnected

adras, 2024-05-15:
### Bambu Studio Version

1.8.2.56

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Connect your lan mode only printer in bambu studio
2. Turn the printer off, wait until BambuStudio notices that it's disconnected
3. Click on the printer to connect to it again (it's still turned off)

### Actual results

4. After the "Connect 3DP-00M-185 failed! [SN:00M09A360201185, code=-1]" message BambuStudio forgot the printer's password

### Expected results

BambuStudio should not forget the printer's password just because the printer is turned off and a connection is attempted

### Project file & Debug log uploads

File size too big: 25 MB are allowed, 37 MB were attempted to upload.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2024-05-17:
Hi. @adras 
Please use our latest version 1.9.1. This issue has been optimized. Thank you.

adras, 2024-05-18:
I just tested it in 1.9.1.66 where the issue still exists. Is there more information about this?

walterwongbbl, 2024-05-21:
> Author

Hi. You need to uninstall the old version first and delete the configuration file at the same time. Do not use overlay installation.

mrtwebdesign, 2024-06-19:
The issue is "optimized"?

I'm on Bambu Studio 1.9.2.57.

I have problems with BS forgetting my P1S password without either the printer or the app being restarted. It feels like it's on a 12 hour timeout.

Where is the configuration file on Macos?
Also, wouldn't this wipe out all of my customizations? Filament profiles? The nuke-it-from-orbit approach feels like overkill. For now I just keep the Bambu password where I can copy-paste it into Bambu studio. I may see if I can automate that process.

I switched to LAN only the last time Bambu cloud services went down for a few hours. I had nothing but time on my hands to find a way to print without the cloud. Other than the inconvenience of Bambu Studio "forgetting" my P1S password, I don't see a reason to go back for my use case.

walterwongbbl, 2024-08-08:
Hi @mrtwebdesign 
I'm very sorry for replying to you so late, you can access the configuration file directory by opening it in the studio menu.
Have you encountered this problem recently.

patrick-at-qreis-dot-com, 2024-09-17:
I can confirm the issue is still present.

I'm using BambuStudio on macOS 15.0. After switching to LAN mode, BambuStudio keeps 'forgetting' the password.

Entering the correct password the printer will connect, but BambuStudio will automatically overwrite the access code / file (BambuStudio.conf, line 3) with an old one (maybe the cloud service one?) or random one (but same one) over and over again.

Quick solution:
Manually changing the code to the correct one and then locking the file will actually keep BambuStudio from 'forgetting' the correct code, but will result in additional files being written (e.g. BambuStudio.conf.54433) that does not seem to do anything(?). 

Works for now as a quickfix, but it is really inconvinient.

Cheers

wolkstein, 2024-09-21:
The same applies to Bambu Studio 1.9.3.50 Linux “Bambu_Studio_linux_ubuntu_24.04_v01.09.03.50.Appimage.
A printer in LAN mode will disconnect from the printer, for example if the computer goes into sleep mode while the Bambo printer is still printing. After the computer wakes up, Bambo Studio will be disconnected from the printer and will be asked for the access code when you reconnect. I now know my code by heart. But it would be nicer if the code was saved for all connection options. So even if the computer is connected to the network via another interface such as WiFi.

sytem, 2024-09-26:
Same issues here with Mac, Version 1.9.5.51 , fresh install few weeks ago.

ansonl, 2024-09-26:
This is still an issue on the latest Windows version 1.9.7.52

adras, 2024-10-04:
Every time my notebook goes into standby I have to reenter the password. It seems like every time the connection is lost the password goes along with it.
This goes up to a point where ftp upload is also broken. I cannot print from my notebook anymore now because I always get timeouts.

phrxmd, 2024-10-12:
I see the same with Linux version 1.9.7.52 and an X1C every time I switch off the printer, when I exit and reopen BambuStudio, or when the computer goes to sleep.

phrxmd, 2024-10-12:
I see the same with Linux version 1.9.7.52 and an X1C every time I switch off the printer, when I exit and reopen BambuStudio, or when the computer goes to sleep.

The workaround (from [this comment](https://github.com/bambulab/BambuStudio/issues/4713#issuecomment-2337542210) on #4713) is to find the configuration file `BambuStudio.conf` (under Linux it's in `~/.config/BambuStudio`, on the Mac it's in `~/Library/Application Support/BambuStudio`). It's a JSON file that should contain entries like this:

```json
{
   "access_code": {
        "00Mxxxxxxxxxxxx": "some_access_code"
    },
    "user_access_code": {
        "00Mxxxxxxxxxxxx": "some_other_access_code"
    },
}
```
It seems that the entries under `access_code` and `user_access_code` go out of sync. When I set both to the value under `user_access_code`, the problem disappears.

herrdeh, 2024-11-01:
A clean installed v 1.10.0.81 on ubuntu Linux has got the same issue.

crashkopf, 2024-11-11:
I was getting annoyed by this same issue.  I'm using Bambu_Studio_ubuntu-v01.09.07.52-24.04.AppImage The access code is being stored properly in the configuration file.

Looking at the code in master we can see that in `GUI_App::init_networking_callbacks()` the app deletes the access code when a connection fails or is lost:
https://github.com/bambulab/BambuStudio/blob/733531b1c68e755da991c9503a09c2206c2e4984/src/slic3r/GUI/GUI_App.cpp#L2070-L2073
https://github.com/bambulab/BambuStudio/blob/733531b1c68e755da991c9503a09c2206c2e4984/src/slic3r/GUI/GUI_App.cpp#L2083-L2087
I hesitate to call it a bug, as it seems to have some intent, but it's definitely not what I would call expected behavior.

skintigh, 2024-12-18:
I can confirm the problem still hasn't gone away by being ignored, maybe you ought to try fixing it. EVERY time my laptop goes to sleep during a print, BS wants me to type in the printer's password. EVERY time BS stops my camera for some reason, when I try to restart it BS asks for the password again. So in this 2 hour print I've had to type it in 10 times. Lets call that 30+times a day, x 365 days a year = for love of god get off your rear and fix this BS. At least include the rock-bottom-basic option of letting BS remember the password. It's insane that you try to force every file to pass through Chinese servers on the way to the printer next to my computer, but then to demand a password 30 times a day for "security." End this BS.

skintigh, 2024-12-18:
I should mention this is LAN-only mode. So I'm not even sure who you think you're protecting me from by forcing me to reenter the password into BS 30 times a day.

skintigh, 2024-12-18:
Even switching tabs makes BS forget which of my ONLY printer I am using, and defaults to none. So EVERY time I change tabs in BS, I have to remind it "hey, BS, you know that one printer I own and no others? The one I was using 10 seconds ago? The one you just sent a print to and is currently running? The one where I WAS watching the live view? Yeah, that one, the ONLY printer, that's the one I would like to use. Just like when you asked my 5 minutes ago, and 5 minutes before that, and 5 minutes before that, and 5 minutes..."

adras, 2024-12-18:
> Even switching tabs makes BS forget which of my ONLY printer I am using, and defaults to none. So EVERY time I change tabs in BS, I have to remind it "hey, BS, you know that one printer I own and no [...]

I'm close to knowing my access key out of my head, given the amount of time this has been ignored so far.

wydmynd, 2024-12-19:
i printed out stickers with my code...

wydmynd, 2024-12-22:
> Hi @mrtwebdesign I'm very sorry for replying to you so late, you can access the configuration file directory by opening it in the studio menu. Have you encountered this problem recently.

your instructions are not clear.
how do I access the  configuration file directory? which menu should I open?
in
C:\Users\%username%\AppData\Roaming\BambuStudio\system\BBL\process
there are many JSON files. Please specify which files / folders to delete
Thanks!

adras, 2024-12-22:
@wydmynd The config file you're looking for should be here: c:\Users\username\AppData\Roaming\BambuStudio\BambuStudio.conf

But I'm not sure how modyfing it in any way would help with the fact that the password is cleared everytime there's an failed connection attempt.

mrtwebdesign, 2024-12-23:
I thought I'd revisit this. If the network password was deleted on purpose for a LAN-only printer connection, it feels a bit user-hostile. I'm a programmer, so I have to deal with network connections all the time. The correct thing to do if a connection is dropped, is to retry, often several times. Deleting a network password, access key or whatever on a connection failure runs against everything I do and the automatic reconnection behavior of other systems I work with.

The "right" way for us to print is to send files halfway around the world through Bambu's servers. This has value to Bambu. Not fixing this is friction to enforce the behavior they want, printing through the cloud. I do not believe anymore that they will fix it. I wonder if industrial customers have this same issue?

I have gotten very good at putting my mouse pointer exactly where the OK button for "wrong password" will show up. I have my bambu printer access code memorized. I have NO motivation to return to using the cloud unless I an understand what Bambu gets out of it.

I love the printer, it works great! But like many things, enjoying the product isn't the same and being happy with the company or their support.

@phrxmd I am going to try the workaround you posted. The location and filename on mac were helpful. How is the world does some number I've never entered end up in there?

jon-walton, 2024-12-23:
I agree, it should be retrying.

While I mainly print via cloud so I can use the mobile app, it is often slow and I wish lan mode was working better

adras, 2024-12-23:
Thanks @mrtwebdesign for clarification. I see it exactly like you from a technical standpoint, I was just looking for another oppinion.
I don't think bambulab has any benefit of this drawing people to the cloud. At least not in my oppinion, because this project is forked quicker than humanly imaginable, since it's already done. I also think, that the other project might be able to fix this rather quickly. And if really everything fails, I just fix it for myself. It's not like they release a new version every couple of weeks :P

ConnectionStatus.Failed and Lost - doesn't sound like a reason to me to forget a password. I could argue that in a case of a bruteforce attack it could make sense to lose the password, e.g ConnectionStatus.InvalidPassword, but even in this case it's entirely useless since a new password is going to be attempted regardless anyway.
Ah, there it is. So you've got the password in BambuStudio, you sell your printer, because the connection is invalid now, the opportunity is taken to also delete it. Especially given possible cloud integration etc...
On the other hand, one just has to save the printers password somewhere else before selling it.

I'm out of reasoning, I tried the best I could to justify it. But, it, doesn't, make, any, sense.

So can we just drop the two lines that delete the stuff on a branch, give it some testing, and then do a PR?

adras, 2024-12-23:
Could we also create kind of a release .zip so other people can test it without building themselves?

crashkopf, 2024-12-23:
> So can we just drop the two lines that delete the stuff on a branch, give it some testing, and then do a PR?

I'd try something like this:
```
} else if (state == ConnectStatus::ConnectStatusFailed) {
                                // obj->set_access_code("");
                                // obj->erase_user_access_code();
                                wxString text;
                                if (msg == "5") {
                                    obj->set_access_code("");   // <-- Moved down here
                                    obj->erase_user_access_code(); // <-- Moved down here                                  
                                    m_device_manager->set_selected_machine("", true);
                                    text = wxString::Format(_L("Incorrect password"));
                                    wxGetApp().show_dialog(text);
                                } else {
                                    text = wxString::Format(_L("Connect %s failed! [SN:%s, code=%s]"), from_u8(obj->dev_name), obj->dev_id, msg);
                                    wxGetApp().show_dialog(text);
                                }
                                event.SetInt(-1);
                            } else if (state == ConnectStatus::ConnectStatusLost) {
                                // obj->set_access_code("");
                                // obj->erase_user_access_code();
                                m_device_manager->localMachineList.erase(obj->dev_id);
                                m_device_manager->set_selected_machine("", true);
                                event.SetInt(-1);
                                BOOST_LOG_TRIVIAL(info) << "set_on_local_connect_fn: state = lost";
                            }
```

skintigh, 2024-12-25:
Bambu will reject you fix if they did this intentionally to punish users for using LAN-only mode an to harass them into using the cloud.

But it shouldn't be hard to modify the binary to jump over those commands and then load it locally.

There seem to be several projects online for jailbreaking the X1C and modifying the firmware. Hopefully they've already fixed this since Bambu won't.

adras, 2024-12-25:
Bambulab did already a fix related to this, however it wasn't enough. Unfortunatelly I cannot get the build environment setup that easily on windows. Otherwise I'd give the fix a try

crashkopf, 2024-12-25:
> Bambu will reject you fix if they did this intentionally to punish users for using LAN-only mode an to harass them into using the cloud.

I doubt there is any intent of punishing users.  The most likely explanation is the function worked "good enough", and there were other fires to be put out.  If you look at the git blame history for those lines you can see they were all added to fix some other issue. 

A slicer is a complicated piece of software with lots of features to maintain.  For most companies any work that doesn't directly translate to more sales is a loss, and having programmers spend time working on small improvements to appease a minority of users isn't likely to get anybody in management excited.

Maybe if we can offer @walterwongbbl our praise and gratitude for spending their time on helping us they will feel more motivated to fix this rough edge in the UI that has us all so vexed.  加油 :grin:

skintigh, 2024-12-26:
I don't know how to read all the notes in the code so I'll take your word for it.

The only thing I know for sure is the complete silence from Bambu on every issue I've run into and googled, and each has been open 1 to 3 years. Not one word from Bambu.

vvzvlad, 2024-12-29:
It's gotten awfully uncomfortable, I agree. I used to enter the code once every six months

VladekRo, 2024-12-31:
> I doubt there is any intent of punishing users. The most likely explanation is the function worked "good enough", and there were other fires to be put out. If you look at the git blame history for [...]

'Minorities' should not be discriminated or forgoten. 😭

crashkopf, 2025-01-01:
> 'Minorities' should not be discriminated or forgoten. 😭
I agree in principle.  I believe in being responsible for what you bring into the world.  Unfortunately, Bambu Studios is a business and plays by the rules of a business.

The only viable alternative slicer I'm aware of with support of Bambu printers is [OrcaSlicer](https://github.com/SoftFever/OrcaSlicer).  It's a community-maintained port of Bambu Studio.  You get all the freedom and no expectation of support.  :laughing:

VladekRo, 2025-01-01:
Thanks for the tip. Orca Slicer do me my P1P and Prusa MK3.
Best Regards.
PS. Also majorities should not be punished by the fault of minorities (I call it a 'Speed Bump Syndrom').

Growing Old is Compulsory. Growing Up is Optional.

Democracy: Where two idiots outvote a genius???

crashkopf, 2025-01-02:
Well folks, I decided to dive in and try for a quick fix.  Initial results seem promising, but all my machines are Linux, so I have no way to test functionality on Windows or MacOS.

VladekRo, 2025-01-02:
Hi, it seems to sufer from the same the same 'dementia' syndrome as BambuLab. Forgets Access Code if p[rinter is swityched off and on.
Best Regards.
Vlad

[cid:9fd77299-5f7c-4371-a2c2-4a3e19616b85]

Growing Old is Compulsory. Growing Up is Optional.

Democracy: Where two idiots outvote a genius???

crashkopf, 2025-01-02:
> Hi, it seems to sufer from the same the same 'dementia' syndrome as BambuLab. Forgets Access Code if p[rinter is swityched off and on.

I only tested it by turning the wifi on my computer off and on.  It shouldn't make a difference though.

Are you building BambuStudio from the  [source code commit](https://github.com/crashkopf/BambuStudio/commit/ac2607044aebe32b2e54855724ca524382adc39c) that I posted?  The changes I made won't be part of the official source code unless approved by Bambu Labs.

VladekRo, 2025-01-03:
Hi, nice piece of code.
No building, I just straight installing, W10.
Sadly, since I finish my Comp Sci Hons in 1995, I did not return to C/C++.
Now I am running out of time, so "Nothing really matter ..."

Growing Old is Compulsory. Growing Up is Optional.

Democracy: Where two idiots outvote a genius???

Swordancer, 2025-02-19:
I solved this issue by using a PowerShell script and changing the shortcut for BambuStudio to the path below on Windows 11. This bug was so frustrating that I don’t think I’ll buy another printer from them. Additionally, I can’t connect the printer to the cloud because my WAN IP is private, or my ISP might be blocking something that the cloud service requires for connection. It works with a mobile network (which has a public IP), so I guess I’m stuck with LAN mode for now. When somone is using LAN modeafter a while, this can get quite annoying. Personally, I've lost my patience with this.

Go to your bambu windows shortcut location C:\ProgramData\Microsoft\Windows\Start Menu\Programs\Bambu Studio or just create new one.

Type this as app location.

C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -ExecutionPolicy Bypass -File "C:\bambu_dementia_solution.ps1"

Create script bambu_dementia_solution.ps1 under C:\ or if you use other location just change required paths and use this content. 

Make copy of your conf file just in case before using it.

`# Function to sanitize non-JSON content (comments) from a file
function Sanitize-ConfFile {
    param (
        [string]$FilePath
    )
    
    # Read the file, remove comments, and save back
    $content = Get-Content $FilePath -Raw
    $sanitizedContent = $content -replace "(#.*$)|(//.*$)", ""  # Removes single-line comments
    Set-Content -Path $FilePath -Value $sanitizedContent
}

# Paths to BambuStudio and BambuStudioBeta .conf files
$confPaths = @(
    "$env:APPDATA\BambuStudio\BambuStudio.conf",
    "$env:APPDATA\BambuStudioBeta\BambuStudio.conf"
)

# Sanitize both config files
foreach ($path in $confPaths) {
    if (Test-Path $path) {
        Sanitize-ConfFile -FilePath $path
    }
}

# Safely read and modify the sanitized JSON content
foreach ($path in $confPaths) {
    if (Test-Path $path) {
        try {
            $Config = Get-Content $path -Raw | ConvertFrom-Json
            if (-not $Config.PSObject.Properties['user_access_code']) {
                $Config | Add-Member -MemberType NoteProperty -Name 'user_access_code' -Value @{}
            }
            $Config.user_access_code["01P00C472401180"] = "38494474"
            $Config | ConvertTo-Json -Depth 3 | Set-Content $path
            Write-Host "Configuration updated in: $path"
        } catch {
            Write-Host "ERROR: Failed to process JSON in $path"
        }
    }
}

Start-Process "C:\Program Files\Bambu Studio\bambu-studio.exe"`

Swordancer, 2025-02-19:
I solved this issue by using a PowerShell script and changing the shortcut for BambuStudio to the path below on Windows 11. This bug was so frustrating that I don’t think I’ll buy another printer from them. Additionally, I can’t connect the printer to the cloud because my WAN IP is private, or my ISP might be blocking something that the cloud service requires for connection. It works with a mobile network (which has a public IP), so I guess I’m stuck with LAN mode for now. When somone is using LAN mode then after a while, this can get quite annoying. Personally, I've lost my patience with this.

Go to your bambu windows shortcut location C:\ProgramData\Microsoft\Windows\Start Menu\Programs\Bambu Studio or just create new one.

Type this as app location.

C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -ExecutionPolicy Bypass -File "C:\bambu_dementia_solution.ps1"

Create script bambu_dementia_solution.ps1 under C:\ or if you use other location just change required paths and use this content. 

Make copy of your conf file just in case before using it.

```
# Function to sanitize non-JSON content (comments) from a file
function Sanitize-ConfFile {
    param (
        [string]$FilePath
    )
    
    # Read the file, remove comments, and save back
    $content = Get-Content $FilePath -Raw
    $sanitizedContent = $content -replace "(#.*$)|(//.*$)", ""  # Removes single-line comments
    Set-Content -Path $FilePath -Value $sanitizedContent
}

# Paths to BambuStudio and BambuStudioBeta .conf files
$confPaths = @(
    "$env:APPDATA\BambuStudio\BambuStudio.conf",
    "$env:APPDATA\BambuStudioBeta\BambuStudio.conf"
)

# Sanitize both config files
foreach ($path in $confPaths) {
    if (Test-Path $path) {
        Sanitize-ConfFile -FilePath $path
    }
}

# Now you can safely read and modify the sanitized JSON content
foreach ($path in $confPaths) {
    if (Test-Path $path) {
        try {
            $Config = Get-Content $path -Raw | ConvertFrom-Json
            if (-not $Config.PSObject.Properties['user_access_code']) {
                $Config | Add-Member -MemberType NoteProperty -Name 'user_access_code' -Value @{}
            }
            $Config.user_access_code["01P00C472401180"] = "38494474"
            $Config | ConvertTo-Json -Depth 3 | Set-Content $path
            Write-Host "Configuration updated in: $path"
        } catch {
            Write-Host "ERROR: Failed to process JSON in $path"
        }
    }
}

Start-Process "C:\Program Files\Bambu Studio\bambu-studio.exe"
```

NVNDO, 2025-03-18:
> bambu_dementia_solution 😂😂😂

VladekRo, 2025-03-19:
v1.10.2.76 seems to remember and reconnect