bambulab/BambuStudio #4077: Bambu Studio 1.9.1.67 on macOS Breaks Z height limit, limit now less than 256mm, breaks auto arrange and printing

FyrbyAdditive, 2024-05-15:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

14.4.1

### Additional system information

MacBook Pro M1 Pro 32GB

### Printer

X1CC

### How to reproduce

- Put model in slicer
- The Z height of the largest part is 250mm, 4mm below the max advertised Z height
- Slicer refuses to auto arrange model onto a plate
- Manually auto arrange model onto plate
- Slice
- Error appears: "A G-Code path goes beyond the max height limit"
- The print button is greyed out
- Printer now does not perform within the specifications on your website (big problem)

### Actual results

Auto arranging broken
Slicing broken
Printer unable to print valid model

### Expected results

Auto arranging works
Slicing works
Able to print valid model

### Project file & Debug log uploads

[60cm Tube Light Shade.3mf.zip](https://github.com/bambulab/BambuStudio/files/15323444/60cm.Tube.Light.Shade.3mf.zip)
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

FyrbyAdditive, 2024-05-15:
I would like to add that this seems to have been a transient failure - I have now selected a different nozzle size and changed back and it seems to work. So this may or may not be a general problem.

FyrbyAdditive, 2024-05-15:
[log.zip](https://github.com/bambulab/BambuStudio/files/15323681/log.zip)

Please find these logs attached, which may be useful in light of the temporary nature of this issue, thank you.

lanewei120, 2024-05-16:
Hi @FyrbyAdditive 

Please note that we have used 250mm for the Z axis limit to ensure there is no damage to the printer during use, due to the following items: 
1. The heatbed might be loosened manually by adjusting the heatbed leveling knobs, changing the Z axis distance to the nozzle. 
2. The different plates used by the printer could have slightly different thickness which can also impact the Z axis distance to the nozzle.
3. If there is a small amount of debris on the base of the printer, the machine could be damaged when the full 256mm volume of the machine is used. 
4. We have also encountered situations where an aftermarket nozzle was slightly longer generating issues when the full Z axis was used. 

To protect the hardware for our customers, we have changed the Z axis limit to account for these small tolerance issues
The maximum printable height of the X1C is limited to 250 in the machine setting section
[image]

Due to the information shared above, we recommend not to change this setting without careful consideration.
We also have a dedicated Wiki article that shows more information about what needs to be done if you want to print with the full 256x256x256 build volume.
https://wiki.bambulab.com/en/knowledge-sharing/print-volume-limitations

lanewei120, 2024-05-17:
@FyrbyAdditive 

could you upload the original 3mf with slicing error to us for a check
the uploaded one doesn't include machine settings

these logic have not been changed for a long time
you can also upload the one which has no issue on previous versions

thanks

FyrbyAdditive, 2024-05-17:
Hi,
Sadly it seems this file is gone, as for this one I copy the 3MF in each time as it doesn't really have any special settings.

I have just tried to re-create this a few times by attempting to carry out the same operations however I am unable to re-create it at this stage, so I suppose if there is no useful data in the log files I may not be able to provide further information.