bambulab/BambuStudio #4078: New Organic Tree Support Issue

Heatmiser89, 2024-05-15:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 10

### Additional system information

Ryzen 7 2700 X
RTX 2080 Super
32 GB Ram

### Printer

A1 Mini

### How to reproduce

Long bridges with Organic Tree Support has does not have proper coverage for support.

https://imgur.com/gallery/support-issue-jHgdZl9

[image]

Images of previous Organic Tree Support results and new Organic Tree Support Results:

https://imgur.com/gallery/support-issue-jHgdZl9

OrcaSlicer organic tree support is correct:

https://imgur.com/gallery/orca-slicer-organic-tree-support-03UbYy7

### Actual results

Using Organic Tree Support does not give correct support coverage. Very thin coverage.

### Expected results

Organic Tree Support should give proper support coverage for bridging. 

### Project file & Debug log uploads

No Zip file. Please see images to support issue present.
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tangketan, 2024-05-16:
Try to set the "max bridge length" option to 0. The logic is, bridges longer than this value will be broken down in Bambu Studio, so setting it to 0 will force the entire bridge to be supported.
I've changed the default value of this option to 0 in future version.

Heatmiser89, 2024-05-16:
> Try to set the "max bridge length" option to 0. The logic is, bridges longer than this value will be broken down in Bambu Studio, so setting it to 0 will force the entire bridge to be supported. I've changed the default value of this option to 0 in future version.

That worked!