bambulab/BambuStudio #4080: Auto-orient for multiple items causes crash

prestonfield, 2024-05-15:
### Bambu Studio Version

1.09.01.66

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10

### Printer

N/A

### How to reproduce

Have multiple objects and auto orient all items on bed

### Actual results

Lag for about 3 seconds then crash

### Expected results

All objects should be auto oriented properly without a crash. 

### Project file & Debug log uploads

[debug_Wed_May_15_13_47_54_7804.log.zip](https://github.com/bambulab/BambuStudio/files/15326514/debug_Wed_May_15_13_47_54_7804.log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tangketan, 2024-05-16:
Please upload the 3mf file for diagnosis.

tangketan, 2024-05-24:
@MrDB42 What you mention is another thing, maybe a feature request, right?
From my understanding, reloading from disk is used when you are unhappy with all the modification/transformation made on the object, and you want to reset everything except the mesh. Therefore, "not re-auto orient" is correct.

MrDB42, 2024-05-24:
@tangketan It showed up in the current release. Prior releases, it worked fine with the example I attached.

Prior, I could create a completely different geometry, with the same name as the already loaded part, select "reload from disk" and the new geometry would come in just fine.