bambulab/BambuStudio #4082: Unable to change fan speed settings in Bambu Studio

saltylemonbreeze, 2024-05-16:
### Bambu Studio Version

1.91.66

### Where is the application from?

Built from source by yourself or others

### OS version

Fedora 40

### Printer

Bambu A1 mini

### How to reproduce

In the prepare window, go to filament settings - cooling. Try to change the fan speed settings on this window.

### Actual results

Not possible to change the fan speed.

Also when I go back to the temperature settings window, now I can't change the temperature any more. Temperature settings only work again after re-starting Bambu studio.

### Expected results

Fan speed settings should be able to choose/edit to change the fan speed.

### Project file & Debug log uploads

[log files.zip](https://github.com/bambulab/BambuStudio/files/15330185/log.files.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

mzient, 2024-05-17:
Duplicate of #3796 ?

kbokh, 2024-05-17:
The same Arch (EndevourOS)
Mostly all settings in the tab Cooling are not changeable. Checkboxes sometimes can be changed, sometimes - not.

mcgiver0510, 2024-05-18:
the same Issue with Arch Linux. Mostly all settings in the tab Filament Settings and Cooling are not changeable.

bambu123, 2024-06-12:
Fixed in version 1.9.2.

closed 2024-06-12 by bambu123