bambulab/BambuStudio #4083: Scarf joint seam error

Mr-121, 2024-05-16:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS14

### Printer

X1C

### How to reproduce

seam: contour
conditional scarf joint: yes
conditional angle threshold: 155

### Actual results

smooth contour has no scarf seam, but contour with sharp corner got scarf seam.

### Expected results

scarf seam on smooth contour rather than contour with sharp corner.

### Project file & Debug log uploads

[Body7.3mf.zip](https://github.com/bambulab/BambuStudio/files/15330268/Body7.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-05-22:
it's a review issue, the seam position is on the end of the loop, but seam pos actualy the start of the loop.  
[image]
[image]
it's a bug, and should be fix later.

QingZhangBambu, 2024-05-22:
the print result should be fine.

Mr-121, 2024-05-22:
Thanks! Hope it can be fixed soon.