bambulab/BambuStudio #4084: Stacking objects is no longer possible. as the right click menu item "Assemble" is missing

christopher-lambe, 2024-05-16:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10

### Printer

P1S

### How to reproduce

https://wiki.bambulab.com/en/software/bambu-studio/stacking-objects

^^ Wiki method of stacking objects (changing the Z coordinate) does not work with version 1.9.1.66

https://wiki.bambulab.com/en/software/bambu-studio/release/release-note-1-9-1
 
^^ The release notes do not mention that the right click menu option is gone.

### Actual results

No right click menu item "Assemble" appears

[image]

### Expected results

Right click menu item "Assemble" should appear

### Project file & Debug log uploads

[Cylinder.zip](https://github.com/bambulab/BambuStudio/files/15330421/Cylinder.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

MadMax389, 2024-05-16:
The name was changed from Assemble to Merge. Function is the same.

christopher-lambe, 2024-05-16:
Thanks guys. I will close the issue. MadMax389 is correct and it works for me now with "Merge"

closed 2024-05-16 by christopher-lambe