bambulab/BambuStudio #4085: When two printers are connected you cannot choose the K value on one of the two.

djmarkino94, 2024-05-16:
### Bambu Studio Version

01.09.01.67

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 14.5

### Printer

Bambulab X1

### How to reproduce

When two printers are connected, I select one then I go to the 4 AMS filaments and set the K values. Then I go to change the printer at the top left, I go back to the AMS filaments and it doesn't let me select the K values, even if these are stored. I am forced to enter them via the printer display. This bug has been present for a long time, even in older versions of Bambustudio. I attach video.

https://github.com/bambulab/BambuStudio/assets/111192738/699f18f3-7ffc-42da-aa0b-d33c8a9cb362

### Actual results

I have to select the values ​​via the printer display.

### Expected results

I should be able to select the values ​​via bambustudio.

### Project file & Debug log uploads

there is no need to attach a project as it can be done even without uploading one.

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-05-16:
@djmarkino94  Has the other printer been calibrated as well? The pa profile of printer A cannot be used by printer B.

djmarkino94, 2024-05-16:
> @djmarkino94  Has the other printer been calibrated as well? The pa profile of printer A cannot be used by printer B.

obviously both are calibrated

djmarkino94, 2024-05-17:
???

djmarkino94, 2024-05-25:
news?

zhimin-zeng-bambulab, 2024-05-28:
@djmarkino94  We are investigating this issue, but it will take some time

djmarkino94, 2024-05-28:
> @djmarkino94  We are investigating this issue, but it will take some time

Ok thanks so much

djmarkino94, 2024-06-11:
hello, no news for this issue?

zhimin-zeng-bambulab, 2024-06-12:
@djmarkino94  We have tried to reproduce it many times, but without success. Could you reproduce this problem? If so, could you share the log 
file when the problem occurred?

djmarkino94, 2024-06-12:
> @djmarkino94 We have tried to reproduce it many times, but without success. Could you reproduce this problem? If so, could you share the log file when the problem occurred?

[Archivio.zip](https://github.com/user-attachments/files/15809796/Archivio.zip)

its' 2 times error, I have try 2 times. same error

zhimin-zeng-bambulab, 2024-06-13:
[image]

Could you try switching to a printer that cannot be display PA profiles normally, and then add a new calibration in the "manage results" of the calibration page to see if it can work normally after adding it.

djmarkino94, 2024-06-13:
I tried but nothing changes. however, the printer that you don't see the PAs is random. if I connect the two printers at the same time, the first one where I choose the PA works, when I switch to the other it doesn't let me choose the PAs

zhimin-zeng-bambulab, 2024-06-14:
1. Could you enable trace service level? You can modify the configuration file as shown below.
[image]
[image]

2. Switch printers on the calibration page instead of the device page.
  After switching the printer, check whether the PA profile can be displayed normally in the manage results dialog.
[image]
[image]

3.  If it cannot be displayed normally, please upload the log of your operation.

Thanks!

djmarkino94, 2024-06-14:
We don't understand each other. I can see the PA values ​​from that calibration page. except that on the devices page I can only set them to one printer, the first one I select. to set them on the other I have to do it from the printer display. I have Mac OS not windows.

zhimin-zeng-bambulab, 2024-06-14:
Let me confirm again. When you switch printer A to printer B on the device page, A can display the PA profile normally, but B cannot, and B can see all the PA profiles(of B) in the "manage results" of the calibration page. B only cannot display PA profile on the device page.

If it is a mac system, you can find the BambuStudio.conf file in the following way.
[image]
[image]

Could you enable trace and then upload the log file if this problem occurs again?

djmarkino94, 2024-06-14:
Exactly, but the thing is random, in the sense that if on the devices page I select printer B first then I can see the PA profiles of B. if I change and go to printer A I don't see the PA profiles of A. So the behavior is the same based on which one I choose first on the devices page. So do I enable the track and try again? If so, do I reload a log of the registry file again?

zhimin-zeng-bambulab, 2024-06-14:
No need to reload a log of the registry file again. After changing to trace,  restart Bambu Studio, a new log file will be generated.  

1.  Switch printers until your "Device" page cannot display the PA profile,
2. Switch to "Calibration" page, and view the PA profile in the "manage results" of the calibration page,
3. Switch to the "Device" page to view the PA profile again.
4. Upload the log file.

zhimin-zeng-bambulab, 2024-06-17:
@djmarkino94  We have tried to make modifications where problems may occur.  Could you try the version below to verify whether the problem has been solved?

https://www.dropbox.com/scl/fi/y4xpkrttrhx4xf4gh9567/Bambu_Studio_mac-v01.09.01.67-20240617112631.dmg?rlkey=nrdwiw1whml3y0mny91cfsw3r&st=tjwr83em&dl=0

djmarkino94, 2024-06-17:
> @djmarkino94 Abbiamo provato ad apportare modifiche laddove potrebbero verificarsi problemi. Potresti provare la versione seguente per verificare se il problema è stato risolto?
> 
> https://www.dropbox.com/scl/fi/y4xpkrttrhx4xf4gh9567/Bambu_Studio_mac-v01.09.01.67-20240617112631.dmg?rlkey=nrdwiw1whml3y0mny91cfsw3r&st=tjwr83em&dl=0

Ok today i try it and and I'll let you know

djmarkino94, 2024-06-17:
Same problem mate... here the log file
[Archivio.zip](https://github.com/user-attachments/files/15866470/Archivio.zip)

zhimin-zeng-bambulab, 2024-06-17:
@djmarkino94   Could you also share the printer log?

djmarkino94, 2024-06-17:
> @djmarkino94   Could you also share the printer log?

Yes tonight i send

djmarkino94, 2024-06-17:
Hello I attach the logs of both printers.

it's a link to my google drive, I didn't know how to send them.

https://drive.google.com/drive/folders/1frlUQaoWEGfYMHS_jCT1wCNTkfVTq7eA?usp=sharing

djmarkino94, 2024-06-19:
the file its ok?

zhimin-zeng-bambulab, 2024-06-20:
No problem was found through the printer logs. I will give another internal version and record more logs to locate the problem.

zhimin-zeng-bambulab, 2024-06-20:
[image]
[image]

No problem were found through the printer logs.
Could you confirm again whether your configuration file has been changed to "trace"? You need to close Studio when modifying this configuration file.

djmarkino94, 2024-06-20:
Yes I confirm mate 💪

djmarkino94, 2024-06-20:
however I have read more than one person having this problem on the Facebook page

zhimin-zeng-bambulab, 2024-06-20:
I will give you an internal version later and will record more logs to help locate the problem.

djmarkino94, 2024-06-20:
Ok mate thanks so much

zhimin-zeng-bambulab, 2024-06-20:
https://www.dropbox.com/scl/fi/iq8h7m6twws612msaj10m/Bambu_Studio_mac-v01.09.01.67-20240620175412.dmg?rlkey=5enpr2w0zamamdczsnxpiofz7&st=w7xb7783&dl=0

Could you reproduce the problem with this version?

djmarkino94, 2024-06-20:
Yes i try this evening

djmarkino94, 2024-06-24:
> https://www.dropbox.com/scl/fi/iq8h7m6twws612msaj10m/Bambu_Studio_mac-v01.09.01.67-20240620175412.dmg?rlkey=5enpr2w0zamamdczsnxpiofz7&st=w7xb7783&dl=0
> 
> Could you reproduce the problem with this version?

Hello mate sorry for late reply, finally which this version the problem it's solved. I tried the new update 1.9.3.50 but the problem occurs with this version.

zhimin-zeng-bambulab, 2024-06-25:
Does the version I gave you solve the problem?

djmarkino94, 2024-06-25:
> Does the version I gave you solve the problem?

Yes mate I have try to simulate the problem 3/4 times but he didn't show up.

djmarkino94, 2024-06-28:
Hello news?

zhimin-zeng-bambulab, 2024-06-28:
The next new version will bring this change. You can use this temporary version to avoid the current problem.

djmarkino94, 2024-06-28:
Thanks so much

djmarkino94, 2024-08-27:
Hi, why hasn't the bug we talked about in the past months been fixed?

zhimin-zeng-bambulab, 2024-08-27:
We added it to the 1.10 beta1 version, but not to the 1.9.X version. Please wait patiently for a while, thank you!

djmarkino94, 2024-08-27:
ok thanks so much

djmarkino94, 2024-10-21:
Hello we have check on beta 1.10 and bug 4085 it's not included, why?

zhimin-zeng-bambulab, 2024-10-21:
The fix for this bug has been merged, but not be specifically mentioned in the release notes. You can have a try in 1.10 Beta1.

djmarkino94, 2024-10-21:
> The fix for this bug has been merged, but not be specifically mentioned in the release notes. You can have a try in 1.10 Beta1.

Ok thanks

closed 2024-10-21 by zhimin-zeng-bambulab