bambulab/BambuStudio #4086: filaments does not change in AMS in 1.9.1

devartmaker, 2024-05-16:
### Bambu Studio Version

1.9.1

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. Installed Bambu Lab Studio version 1.9.1
2. Two-Color Modeling Output Box

### Actual results

1. After installing Bamboo Studio version 1.9.1, the filament does not change in AMS.
When changing the filament, the same filament keeps spraying in front and the output continues as shown in the attached video.

2. the power fan noise is too loud when preparing to print.

https://github.com/bambulab/BambuStudio/assets/40697700/a9055642-976a-4092-8e80-0cdfd36d3c74

### Expected results

After reinstalling the 1.8.4 version, the output is normal.

### Project file & Debug log uploads

[test.zip](https://github.com/bambulab/BambuStudio/files/15333917/test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-05-16:
@devartmaker  Is it use the incorrect filament? And could you share your gcode file?

devartmaker, 2024-05-16:
> @devartmaker Is it use the incorrect filament? And could you share your gcode file?

I use e-sun filament.
I'll add a gcode file that I made in 1.9.1
[test_PLA_33m23s.zip](https://github.com/bambulab/BambuStudio/files/15337044/test_PLA_33m23s.zip)