bambulab/BambuStudio #4087: Ability to adjust X1C camera exposure, or change how the auto adjsut work

capsel22, 2024-05-16:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab github releases

### OS version

Win11

### Printer

X1C

### How to reproduce

1. Turn on top mounted LED lights
2. X1C camera over adjusts so the model looks dark
3. Turn off top mounted LED lights
4. X1C camera over adjusts again and the model still looks dark

### Actual results

Model looks dark with the top mounted LED light on or off. the auto adjust mechanism doesn't do a good job picking the correct exposure

https://github.com/bambulab/BambuStudio/assets/38831192/ea2901e3-e037-4600-8235-44a9dd6dffa8

### Expected results

When in "Advanced" or "Developer Mode" allow an extra option to set the exposure MANUALLY with a slider under camera options.
This manual exposure should persist between restarts.

IDEALLY, this camera exposure would be exposed to MQTT so we could use Home Assistant or other automations to adjust the camera exposure based on own criteria.

### Project file & Debug log uploads

video attached above

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-05-16:
the same as https://github.com/bambulab/BambuStudio/issues/4061

capsel22, 2024-05-16:
> the same as #4061

plus with an idea for MQTT integration

CCS86, 2024-07-28:
We need better exposure control!

CCS86, 2024-07-28:
https://github.com/bambulab/BambuStudio/issues/3148

Teedabbeljoe, 2024-12-01:
Would like this too, I can see the camera is technically able to expose correctly when switching on and off the lamp on my A1 mini. However the auto exposure makes it impossible to see the object under every condition (natural light, artificial light, harsh light and soft light)