bambulab/BambuStudio #4088: Rotation bug when rotation a Z or Y mirrored part

PanamonCreel, 2024-05-16:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

WIN10

### Printer

P1S

### How to reproduce

1. Mirror a part on the plate on Z or Y axis
2. Rotate the Mirrored part on any axis, observe part flipping along the axis it was mirrored on initially
3. Rotate the Mirrored part again on any axis, observe part flipping again along the axis it was mirrored on initially 

### Actual results

Part will flip along the initial mirrored axis with each rotation increment no matter which rotation axis was used. So if mirrored along Z it will flip the part along the Z axis when rotated. If mirrored along Y the part will flip along Y when rotated.  When mirrored along X the part will not flip when rotated (as it should be)

### Expected results

a mirrored part should not flip around with each rotation operation.

### Project file & Debug log uploads

[RotationBugExample.zip](https://github.com/bambulab/BambuStudio/files/15336660/RotationBugExample.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2024-05-17:
Can you record a gif to describe your problem? Thank you

PanamonCreel, 2024-05-17:
Certainly can do that
[image]

PanamonCreel, 2024-06-12:
Rotation bug still exists with version 1.9.2.57

Haidiye00, 2024-06-13:
Thank you for your GIF. We will investigate the cause as soon as possible

jedthe3rd, 2024-08-22:
Just got my X1C today and I ran into this issue. I am on 1.9.3.50

Haidiye00, 2024-08-22:
The problem has been identified and is currently being tested

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00