bambulab/BambuStudio #4092: Bambu Lab Studio Memory Leak?

ClawzSkunk, 2024-05-16:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: AMD Ryzen 9 3950x
Motherboard: MSI X570 UNIFY
Memory: 32GB DDR4 3200

### Printer

Bambu Lab X1C

### How to reproduce

Every time I try to slice with tree supports, the memory usage jumps up to a crazy level - I've attached 3 different attempts. This always occurs while generating the G-Code Layers
[image]
[image]
[image]

### Actual results

As mentioned memory usage goes to mad heights and the application crashes whenever I try to slice. 
[image]
[image]
[image]

### Expected results

Slicing should complete as normal. 

### Project file & Debug log uploads

Exceptions:

Exception Code :c0000005 ACCESS_VIOLATION

Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0xffffffff 
Context :0000000F684E8800 
ContextFlag : 0x10005f, EFlags: 0x10202 
Fault address:  0x6273662D 0x0:0x0 

Show CallStack:

Call SymGetLineFromAddr64 ,Address 6273662d Error:000001e7

Call SymGetLineFromAddr64 ,Address 6276990d Error:000001e7

Call SymGetLineFromAddr64 ,Address 6275be1b Error:000001e7

Call SymGetLineFromAddr64 ,Address 62769351 Error:000001e7

Call SymGetLineFromAddr64 ,Address 629e5390 Error:000001e7

Call SymGetLineFromAddr64 ,Address 629f1dd8 Error:000001e7

Call SymGetLineFromAddr64 ,Address 629e536d Error:000001e7

Call SymGetLineFromAddr64 ,Address 629f1dd8 Error:000001e7
[... 326 lines cut ...]

Call SymGetLineFromAddr64 ,Address 629ee903 Error:000001e7

Call SymGetSymFromAddr64 ,Address 623c04c0 Error:000001e7

Call SymGetSymFromAddr64 ,Address 623bbae2 Error:000001e7

Call SymGetSymFromAddr64 ,Address 623c6e3c Error:000001e7

Call SymGetSymFromAddr64 ,Address 623ce307 Error:000001e7

Call SymGetSymFromAddr64 ,Address 623ce1cb Error:000001e7

Call SymGetLineFromAddr64 ,Address d43b1bb2 Error:000001e7

Call SymGetLineFromAddr64 ,Address d6507344 Error:000001e7

Call SymGetLineFromAddr64 ,Address d66426b1 Error:000001e7

[MagSafe_Head_Entire.zip](https://github.com/bambulab/BambuStudio/files/15341811/MagSafe_Head_Entire.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

odysoftware, 2024-05-22:
Having same issue 1.9.1.66 crashing a lot on complex geometry with support enabled :(

ClawzSkunk, 2024-05-22:
> Having same issue 1.9.1.66 crashing a lot on complex geometry with support enabled :(

Mine was with supports enabled as well

odysoftware, 2024-05-22:
The bad thing is that i happens randomly but you can't work with larger projects because sooner or later it just closes. I tried to open the same project on the Mac Version 1.9.1.67 and there it works without any issues. Something is wrong here in the windows version.

odysoftware, 2024-05-23:
Just to let you know, I was able to fix it by using the latest Asus BIOS and enabling the option -> Intel Baseline Profile. So if you are having a Intel System with a 13th or 14th gen processor you need to activate the intel baseline profile - as it looks now that completely stopped those access violation errors and Bambu Studio closes :) - I read about instability problems on 13th and 14th gen due to native overclocks of motherboard brands!

I can confirm after that change I can now slice complex geometry and not encountered a single crash regardless of support/options/etc. ! So the source of this error may be related to your system.

ClawzSkunk, 2024-05-23:
Sadly I have an AMD 3950x

odysoftware, 2024-05-23:
Yes but if it happens randomly - then I think it might be related to some bios/overclocking issues?

QingZhangBambu, 2024-06-03:
sorry, haven't yet. I still try to figure out that

sorren4, 2024-06-18:
Mine isn't quite that bad, but mostly seems to bog down my system when left open for hours.  I can understand if it's streaming camera data but in my case I usually have the camera closed so the feedback from the printer on status should be very minimal.  Not sure why it's generating so much network traffic and eating up so much memory after a while.

ClawzSkunk, 2024-07-14:
> sorry, haven't yet. I still try to figure out that

Can you tell me whether you've made any progress on this at all. It's becoming very frustrating even on simple models. 
[image]

Jay-Crypto1, 2025-06-11:
yea I can't get it to clone anything without it crashing I thought it was my computer but seeing how I'm not the only one. Bambu needs to fix this ASAP