bambulab/BambuStudio #4095: Studio 1.9 Creates .3MF Project files without being asked

tanj666, 2024-05-17:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

13th Gen Intel(R) Core(TM) i7-13700KF   3.40 GHz, 64 bit
32Gb RAM
Nvidea Quadro

### Printer

Bambu P1P

### How to reproduce

Start a new project
Open an STL file (such as xxx.stl)  making sure no .3MF file of same name exists (so no xxx.3mf exists)
Slice it
Create a new project
- check filesystem, a file has been created called xxx.3mf
Windows 3D Builder cannot open this file only Bambu Studio can

### Actual results

A new project file is created and saved to the filesystem, none was asked for.  Auto backup has been disabled

### Expected results

A .3mf file should NOT have been created without being requested

This means I cannot use Bambu Studio as it is.  I do not use the project workflow system and never have and do not like my filesystem being filled up with useless files.

### Project file & Debug log uploads

[2Firkin - Big.zip](https://github.com/bambulab/BambuStudio/files/15345571/2Firkin.-.Big.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

lanewei120, 2024-05-17:
@tanj666 
hello

could you give a gif to recording the whole process?

thanks

tanj666, 2024-05-17:
Lane.Wei,
I'm sorry, I don't know your correct form of address as I know that Chinese
names have forename/family name reversed compared to western names.

I cannot create a .GIF as the best I can to recreate the process takes 30
seconds.  Instead I am providing a link to an .MP4 video showing what
happens, there is no audio on this file.

Here's the link:
https://drive.google.com/file/d/1qNwEl06r30x1dz8YSUypyhOZ63mWE7dD/view?usp=drive_link

Regards
Anthony