bambulab/BambuStudio #4097: Problems with KEEP HORIZONTAL checkbox, Clicking Vertical in the Seam Paint UI should not automatically select the View: keep horizontal checkbox

evolutionlures, 2024-05-17:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

Macos 13.6.6

### Printer

Bambu Lab P1P

### How to reproduce

open bambu studio, load a model, click the "Seam Paint" button, hit the hotkey command 4 to rotate the view to the back side, then click the "Vertical" Checkbox and notice that it automatically chooses to turn on "View: horizontal". Watch as the view is now auto rotated to the front view.

### Actual results

The view is abruptly changed to a different direction than what was already chosen.

### Expected results

I am a power user, i use this software many times a day and depend on hotkeys to get my work done. Every product I sell requires a seam to be placed perfectly vertical on the back side of the model. This new feature really breaks the flow for me. 
The first problem is that I even have to paint a seam on the back of these simple cylindrical models. The creation of the paint vertical feature has been a great time saver, but now this newer feature breaks the flow. I need to click command 4, then paint directly vertically on the back side of the model, not the front. Clicking the vertical checkbox should not change the view if it is already vertical, and it should not auto select the checkbox. 

Also a side note, it would be excellent if tab key would auto switch between the sliced view and preview, like orca does. 

### Project file & Debug log uploads

no project file is needed, this is a UI bug

### Checklist of files to include

- [X] Log file
- [ ] Project file

Haidiye00, 2024-05-20:
Understood, we will make improvements based on your suggestions. Thank you

Haidiye00, 2024-05-20:
[image]
Is this what you expected

evolutionlures, 2024-05-20:
Not really I don't understand why the checkbox "keep view horizontal" automatically checks on when selecting the vertical checkbox.Why not just let the user turn it on when needed? This new slider in the video adds even more clicks. I already use hotkeys to rotate the view to front or back.Sent from my Galaxy

03.gif (view on web)

Haidiye00, 2024-05-28:
ok，i will cancel "Keep horizontal" linkage selected when click "Vertical"

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00