bambulab/BambuStudio #4098: Slicing Failing with Latest Studio using Bambu Filament; long_retractions_when_cut not a variable

RPINerd, 2024-05-18:
### Bambu Studio Version

1.9.0.60

### Where is the application from?

Bambu Lab github releases

### OS version

Linux Fedora Build

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1- Set object to use official Bambu Labs filament
2- Click `Slice plate`

### Actual results

Error message pops up:
[image]

However the custom G-code has never been changed and it doesn't even have 18 lines:
[image]
And it ONLY happens with Bambu Filament, but again, this error line does not appear in the filament profiles

### Expected results

The file slices

### Project file & Debug log uploads

Error happens on any project, any object, any settings as long as an official Bambu Filament is chosen

There are no log files being generated since updating to 1.9

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-05-20:
the latest is fine, could you download version 1.9.66? and if the latest version is still not work please upload your 3mf

RPINerd, 2024-05-20:
Looks like that was the issue, odd that the repo didn't recognize a new version when installing from git.. anyways, thanks for your help!

closed 2024-05-20 by RPINerd